wpilibsuite/allwpilib #684: Allow receiving CAN packets without exceptions being thrown when no messages are available

WasabiFan, 2017-10-25:
Currently, the C++ signature for the low-level CAN receive method looks like this:

https://github.com/wpilibsuite/allwpilib/blob/a6e6ae41b95caf45c2fd7a1ed4dde5420b31802b/ni-libraries/include/FRC_NetworkCommunication/CANSessionMux.h#L56

Notice that it "returns" a status code. However, in the mapping to the Java interface, the behavior was changed a bit:

https://github.com/wpilibsuite/allwpilib/blob/a6e6ae41b95caf45c2fd7a1ed4dde5420b31802b/wpilibj/src/main/native/cpp/CANJNI.cpp#L83-L126

Note the call to `CheckCANStatus`, which throws a Java exception whenever the status code is nonzero.

My current goal is to interface with a custom CAN device on the bus. In C++, this can be done easily and efficiently [as is done here](https://github.com/wpilibsuite/allwpilib/blob/master/hal/src/main/native/athena/ctre/CtreCanNode.cpp) by calling the receive function and checking the status code. If the status code is `ERR_CANSessionMux_MessageNotFound`, you can wait and check again. However, due to the way the Java interface was implemented, this doesn't work from Java. You must instead catch an Exception every time you attempt a read, which is quite damaging to performance if you intend to read it in your main robot loop.

Could the interface be changed to allow for polling for messages without catching exceptions every time? Or is there another approach I'm missing?

After spending a fair amount of time on this, I was able to interface with the low-level APIs through [JNA](https://github.com/java-native-access/jna) -- thankfully, those APIs were marked with `extern "C"` -- but it would be much nicer to have it built-in so I didn't have to integrate this external library.

ThadHouse, 2017-10-25:
We would definitely want to keep the existing API, as to most Java users, exceptions being thrown on errors is the usual method of detecting errors. Have you actually measured performance issues with catching exceptions? Yes they are slower, but 1 per 20ms code loop shouldn't affect performance noticeably.

The big issue with adding something that checks status is that Java doesn't have an easy way to pass a status variable by reference. We could add an overload that has an IntBuffer overload, but unless the exceptions are noticeably impacting performance,  it might not be necessary to implement. Polling for messages doesn't work in a thread safe manor, so an IntBuffer is really the only way, and it does still mean an allocation for every CAN call.

WasabiFan, 2017-10-25:
I can do some more scientific profiling in the next few days to try to get more concrete numbers. Anecdotally, what I saw when I did this last week was:
- An increase in memory usage reported by the DS
- An increase in CPU usage -- again, reported by the DS IIRC

Both observations were in the case where no CAN messages were being sent by the other device, compared to when many messages were being read. In other words, no messages to be read made the robot code slower.

These were by no means done in a controlled environment so I wouldn't trust those items until I can get some hard data (or demonstrate I did it wrong the first time). Discussions such as [this one](https://stackoverflow.com/questions/299068/how-slow-are-java-exceptions) seem to indicate that there is a significant penalty to instantiating exception objects, but tests like those are done without much other work to put the delay in comparison with other things going on.

WasabiFan, 2017-10-25:
WRT the ideal API question: as a Java developer, the behavior I would expect in this scenario without regard to the technical limitations of JNI would be for the method to return a byte array which was null when no message was available. Or, similarly, it would return a Boolean and populate an output array like a C#-style `TryParse` method. It would throw exceptions only when the bus itself had a communications error. After all, _Exceptions are supposed to be exceptional_, and having no data to read is not exceptional in my mind. I'm finding that the fastest I can get an Arduino to send messages still only results in a message being available one in every ten robot loops.

Perhaps the best solution here if this is indeed a perfect issue would be to add a `tryReceiveMessage` method :grinning:

ThadHouse, 2017-10-25:
That seems odd where you can only get an arduino to send a message every 200ms (Each robot loop is 20ms if you are using IterativeRobot). The CAN bus is definitely faster then that, and I've use arduino CAN before and gotten much better results then that.

We definitely would not be changing the existing behavior of the receive function, as there are people who depend on the current behavior. I don't like the idea of an overload returning null, as there is no way to receive what the error code was. However, we could take an array in, and then just fill that array if valid data was received, and return the error code on failure. Note that this could still throw exceptions because of bad arrays, and it would also be up to the users to make sure the array is used in a thread safe way. But if those are ok restrictions, that is probably the easiest overload to make.

WasabiFan, 2017-10-25:
I'm going to call this "good enough" as-is. After taking another look, in my original testing I was using a very simple app that attempted to run the CAN receive call hundreds of times every robot loop. In this environment, CPU utilization drops an average of 5-10% when messages are being received (from 42% to ~34%). However, in practice, I only expect to send messages at a maximum of 30Hz or 60Hz; this is low enough that the change in behavior is negligible. Memory difference in either case was less than 1mb, so inconsequential for my purposes. If I come across any issues with it I'll re-open the issue.

closed 2017-10-25 by WasabiFan