wpilibsuite/allwpilib #698: Utility.getFPGATime() encounters a native error

JacisNonsense, 2017-10-29:
It seems as though calling `Utility.getFPGATime()` throws an error upon reaching HALUtil.getFPGATime() (native method). Error code is -63194.

```
Error at edu.wpi.first.wpilibj.RobotBase.main(RobotBase.java:256): ERROR Unhandled exception: java.lang.RuntimeException:  Code: -63194.
 Unknown error status at [edu.wpi.first.wpilibj.hal.HALUtil.getFPGATime(Native Method),
edu.wpi.first.wpilibj.Utility.getFPGATime(Utility.java:48), 
edu.wpi.first.wpilibj.internal.HardwareTimer.getFPGATimestamp(HardwareTimer.java:46), 
edu.wpi.first.wpilibj.Timer.getFPGATimestamp(Timer.java:29), 
edu.wpi.first.wpilibj.MotorSafetyHelper.<init>(MotorSafetyHelper.java:38), 
com.ctre.phoenix.MotorControl.SmartMotorController.<init>(SmartMotorController.java:367), 
com.ctre.phoenix.MotorControl.CAN.TalonSRX.<init>(TalonSRX.java:14), 
test.MyClass.robotInit(MyClass.java:10), 
edu.wpi.first.wpilibj.IterativeRobot.startCompetition(IterativeRobot.java:31), 
edu.wpi.first.wpilibj.RobotBase.main(RobotBase.java:254)]
```

PeterJohnson, 2017-10-29:
What roborio image version and wpilib version?  This just changed for 2018 beta2.

JacisNonsense, 2017-10-29:
WPILib 2018.1.1-beta-2
Roborio 2018v9

JacisNonsense, 2017-10-29:
Update: just reimaged and updated firmware and it seems to be all good now, looks like a remnant of my broken installation.

JacisNonsense, 2017-10-29:
Nevermind, issue has started popping up again after a reboot, seems to be unaffected by firmware.

ThadHouse, 2017-10-29:
Are you using ./robotCommand to run this, or frcKillRobot -t -r. Just using ./robotCommand by itself has been known to cause issues like this before.

JacisNonsense, 2017-10-29:
I've been using `./robotCommand`, and it would seem `frcKillRobot -t -r` does fix the issue. Could you give some insight as to why this discrepancy exists?

ThadHouse, 2017-10-29:
The frcKillRobot command sets a whole bunch of environment variables that the fpga sometimes requires. Without them, the fpga either ends up throwing that version error you're seeing, or memory errors. Also, frcKillRobot starts up the netconsole pipe, and enables the DS reboot robot code button. All ./robotCommand is there for is to have a common place for all languages to put what command they need to start.

JacisNonsense, 2017-10-29:
Got it, seems like using the `frcKillRobot.sh` tool is more necessary than in previous years. Thanks for the update

closed 2017-10-29 by JacisNonsense