wpilibsuite/allwpilib #729: std::mutex has priority inversions

AustinSchuh, 2017-11-13:
All classes which use ::std::mutex have priority inversions when used from multiple threads.  This is a massive regression for us.

PeterJohnson, 2017-11-13:
Where do you see this impacting performance?  I would prefer seeing us change the locking approach than use a non-standard mutex.

AustinSchuh, 2017-11-13:
I'll look into it a bit, but any object that I interact with regularly has to be guaranteed to not interact with the ::std::mutex.  I currently count ~60 instances of ::std::mutex.  It'll be very very hard to prove that there will be no cross thread usages of any of those mutexes.  The burden of proof really should be going the other way.  How can you show that a ::std::mutex is not used across threads when a robot is running for all use cases?

MotorSafetyHelper is broken now.  The driver station thread accesses all of the motor objects.  Our DS thread runs at a different priority than our control loop threads.  That means that there's a priority inversion when controlling motors.  That's the first instance I've found, and I'm sure there are more.

This is RT, so it's about worst case performance, not average performance.  Average performance will likely show no issues.  I've been part of enough of user space timing problems at work which were traced down to using a non-pi safe mutex.  I typically solve this type of problem by instrumenting the API in question and measuring when my guarantees are violated.  In this case, I would do that by storing which threads have accessed which mutexes, and blowing up when 2 threads access a mutex.

DMA broke on me, so I need to fix that before I can get any of our code running to run real tests.

It's also very disingenuous to write "hal/src/main/native/include/HAL/cpp/priority_mutex.h:typedef std::mutex priority_mutex"  If you are going to change the semantics of that mutex, you need to change the name.  The contract of that mutex was that it was PI safe, and constexpr.  Please remove that header entirely to force teams to make the hard decision of what to do rather than have their code silently broken.

PeterJohnson, 2017-11-13:
I understand your concerns regarding PI with multiple priority levels, and while this use case is extremely rare in FRC, I can see the value in defaulting to PI-supporting mutexes throughout WPILib.

The reason for the typedef is on non-Linux platforms (namely windows!) which don't have the capability to fix PI and/or don't care about PI.  The "naming" concern of this we'll likely fix by having a "wpi::mutex" that's typedefed appropriately to either wpi::priority_mutex or std::mutex depending on the platform, and then it's indeed the user's problem to handle the cross-platform case if they explicitly use wpi::priority_mutex.

AustinSchuh, 2017-11-13:
Thanks Peter.

I realized after I typed my complaint about the typedef that it was for Windows support, which is understandable (and unfortunate!).  I like your proposed solution of removing the typedef and adding wpi (or frc)::mutex which routes to the correct mutex, and using priority_mutex where it matters.

I need to get DMA working before I could put together a patch, but I'm up for contributing one.

PeterJohnson, 2017-11-13:
PR to add the priority mutex and CV into the wpi namespace is wpilibsuite/wpiutil#50.

Regarding DMA, I was starting to look at your code from last year, but it sounds like I should wait until you update it.  You can either PR it or just email it to me.

closed 2017-11-13 by PeterJohnson

AustinSchuh, 2017-11-13:
Thanks Peter!  I'll start testing this commit.

PR #730 (2017-11-13): Use wpi::mutex instead of std::mutex.
This uses a priority-aware mutex on Linux platforms.

Fixes #729.
commit 4d559f38 (2017-11-13): Use wpi::mutex instead of std::mutex. (#730)

This uses a priority-aware mutex on Linux platforms.

Fixes #729.