wpilibsuite/allwpilib #733: SPI Direct Byte Buffer GC issues

JacisNonsense, 2017-11-14:
This was reported by @JamieSinn as part of the beta testing.

It would seem that `SPI#read` and `SPI#write` are taking up an extraordinary amount of memory and the Direct Byte Buffers within are not being released. 

https://i.imgur.com/PMTpaQC.png
https://i.imgur.com/Se7v6O0.png

The GC may not be triggering, or perhaps the cleaners of the respective DBB's are not being called properly. This doesn't appear to be a memory issue with WPILib, although looking through the Zulu implementation nothing strikes me as particularly odd.

@JamieSinn if you could run this test again manually calling `System#gc` that might provide some further insight.

JamieSinn, 2017-11-15:
@JacisNonsense @PeterJohnson - Here's a screenshot of the call tree immediately post a forced GC.
https://i.imgur.com/oj9ZpHq.png

PeterJohnson, 2017-11-15:
As shown in the above PR, the plan is to completely remove the use of direct ByteBuffers (in particular when they're allocated for each call) to avoid this issue entirely.  I2C and SPI were one of the last places they were being used in WPILib.

JamieSinn, 2017-11-15:
Thanks Peter!

PR #734 (2017-11-17): Switch JNI interfaces from ByteBuffer to byte[]
This avoids a direct byte buffer allocation on every read/write/transaction
for the byte[] variants.

Also change spiGetAccumulatorOutput() to directly set the AccumulatorResult
object, avoiding a ByteBuffer allocation.

Changes HAL SPI interfaces to use const for dataToSend.

Fixes #733.
commit 6307d410 (2017-11-17): SPI: Provide byte[] JNI interfaces.

This avoids a direct byte buffer allocation on every read/write/transaction
for the byte[] variants.

Also change spiGetAccumulatorOutput() to directly set the AccumulatorResult
object, avoiding a ByteBuffer allocation.

Changes HAL SPI interfaces to use const for dataToSend.

Fixes #733.
closed 2017-11-17 by PeterJohnson