wpilibsuite/allwpilib #736: [Issue/Feature Request] Volatile memory not being used

AustinInSeattle, 2017-11-14:
The RoboRIO has 512MB of nonvolatile memory, and part of that is 256MB of DDR3 memory. However, upon reaching 256MB of memory, the FRC_UserProgram.jar will crash.

Tested during the 2017 competition season with WPILib's public release in Eclipse for Zulu JDK1.8.0.

ThadHouse, 2017-11-14:
Confused on the tested during the 2017 competition season part. Did you notice this during the 2017 season with the JDK installed using the Java Installer, or your 2017 code with newer builds. The Zulu JDK is new for 2018, and would not have been used by teams in 2017.

Kevin-OConnor, 2017-11-14:
The roboRIO has 256 MB of RAM (DDR3) and 512 MB of flash storage (non-volatile memory). These two are not interchangeable. What exactly are you asking for here?

AustinInSeattle, 2017-11-16:
Overall, just noticed that the RIO seems to have crashed after 256MB of RAM rather than after reaching a complete OOM (OS-level) error.

Kevin-OConnor, 2017-11-16:
256MB is all the RAM.

closed 2017-11-16 by Kevin-OConnor