wpilibsuite/allwpilib #738: Rename interruptable -> interruptible

Twinters007, 2017-11-15:


karagenit, 2017-11-26:
Hey, a quick `grep` of the repository shows the following occurrences of "interruptable":

```
./wpilibj/src/main/java/edu/wpi/first/wpilibj/command/Scheduler.java:128:   * uninterruptable. Giving <code>null</code> does nothing.
./wpilibjIntegrationTests/src/main/java/edu/wpi/first/wpilibj/AbstractInterruptTest.java:31:  private InterruptableSensorBase m_interruptable = null;
./wpilibjIntegrationTests/src/main/java/edu/wpi/first/wpilibj/AbstractInterruptTest.java:34:    if (m_interruptable == null) {
./wpilibjIntegrationTests/src/main/java/edu/wpi/first/wpilibj/AbstractInterruptTest.java:35:      m_interruptable = giveInterruptableSensorBase();
./wpilibjIntegrationTests/src/main/java/edu/wpi/first/wpilibj/AbstractInterruptTest.java:37:    return m_interruptable;
./wpilibjIntegrationTests/src/main/java/edu/wpi/first/wpilibj/AbstractInterruptTest.java:43:    if (m_interruptable != null) {
./wpilibjIntegrationTests/src/main/java/edu/wpi/first/wpilibj/AbstractInterruptTest.java:45:      m_interruptable = null;
./wpilibjIntegrationTests/src/main/java/edu/wpi/first/wpilibj/AbstractInterruptTest.java:50:   * Give the interruptable sensor base that interrupts can be attached to.
./wpilibjIntegrationTests/src/main/java/edu/wpi/first/wpilibj/AbstractInterruptTest.java:55:   * Cleans up the interruptable sensor base. This is only called if {@link
```

So as far as user-code is concerned, only a bit of JavaDoc in `Scheduler.java` is misspelled. Are there any other instances you noticed?

Twinters007, 2017-11-26:
I guess since it's interruptable it's not terribly important, but still

karagenit, 2017-11-27:
Oh I can change it pretty easily - I just wanted to make sure I wasn't missing any other misspellings before I make a PR. 

Looking at the above instances more closely shows that the Scheduler bit is actually on a private method (`_add`) so it isn't a part of the published JavaDocs (though should probably still be changed). 

*However*, there is actually an entire class that was misspelled: `InterruptableSensorBase`, in both WPILibC and WPILibJ... I'm not sure how renaming this will affect user-code. I'm assuming 99% of teams only ever use the DigitalInput/DigitalOutput subclasses, but in general we like to avoid breaking changes in favor of a deprecation/removal cycle (though this isn't an issue for internal changes, which it seems like this is).

karagenit, 2017-11-29:
Quick follow up, just realized my `grep` didn't include uppercase:

```
./wpilibj/build/reports/checkstyle/athena.html:175:<td><a href="#f-/home/caleb/documents/programming/src/allwpilib/wpilibj/src/athena/java/edu/wpi/first/wpilibj/InterruptableSensorBase.java">/home/caleb/documents/programming/src/allwpilib/wpilibj/src/athena/java/edu/wpi/first/wpilibj/InterruptableSensorBase.java</a></td><td>0</td>
./wpilibj/build/reports/checkstyle/athena.html:614:<a href="#top">Back to top</a><a name="f-/home/caleb/documents/programming/src/allwpilib/wpilibj/src/athena/java/edu/wpi/first/wpilibj/InterruptableSensorBase.java"></a>
./wpilibj/build/reports/checkstyle/athena.html:615:<h3>File /home/caleb/documents/programming/src/allwpilib/wpilibj/src/athena/java/edu/wpi/first/wpilibj/InterruptableSensorBase.java</h3>
./wpilibj/build/reports/checkstyle/athena.xml:63:<file name="/home/caleb/documents/programming/src/allwpilib/wpilibj/src/athena/java/edu/wpi/first/wpilibj/InterruptableSensorBase.java">
./wpilibj/src/main/java/edu/wpi/first/wpilibj/command/CommandGroup.java:338:   * {@link CommandGroup#setInterruptible(boolean) setInterruptable(false)} was called or if it is
./wpilibj/src/main/java/edu/wpi/first/wpilibj/InterruptableSensorBase.java:17:public abstract class InterruptableSensorBase extends SensorBase {
./wpilibj/src/main/java/edu/wpi/first/wpilibj/InterruptableSensorBase.java:44:  public InterruptableSensorBase() {
./wpilibj/src/main/java/edu/wpi/first/wpilibj/DigitalSource.java:16:public abstract class DigitalSource extends InterruptableSensorBase {
./wpilibj/src/main/java/edu/wpi/first/wpilibj/InterruptHandlerFunction.java:49:   * when {@link InterruptableSensorBase#requestInterrupts(InterruptHandlerFunction)} is run.
./wpilibcIntegrationTests/src/FRCUserProgram/cpp/DIOLoopTest.cpp:13:#include "InterruptableSensorBase.h"
./wpilibcIntegrationTests/src/FRCUserProgram/cpp/DIOLoopTest.cpp:85:  InterruptableSensorBase::WaitResult result =
./wpilibcIntegrationTests/src/FRCUserProgram/cpp/DIOLoopTest.cpp:108:  EXPECT_EQ(InterruptableSensorBase::WaitResult::kFallingEdge, result)
./wpilibc/build/tmp/linkWpilibcSharedLibrary/options.txt:36:/home/caleb/documents/programming/src/allwpilib/wpilibc/build/objs/wpilibc/shared/wpilibcCpp/dyj2tpj92ffrfim8vz6rnxf4o/InterruptableSensorBase.o
./wpilibc/build/tmp/compileWpilibcStaticLibraryWpilibcCpp/output.txt:210:compiling InterruptableSensorBase.cpp successful.
./wpilibc/build/tmp/compileWpilibcSharedLibraryWpilibcCpp/output.txt:210:compiling InterruptableSensorBase.cpp successful.
./wpilibc/src/main/native/cpp/InterruptableSensorBase.cpp:8:#include "InterruptableSensorBase.h"
./wpilibc/src/main/native/cpp/InterruptableSensorBase.cpp:25:void InterruptableSensorBase::RequestInterrupts(
./wpilibc/src/main/native/cpp/InterruptableSensorBase.cpp:50:void InterruptableSensorBase::RequestInterrupts() {
./wpilibc/src/main/native/cpp/InterruptableSensorBase.cpp:66:void InterruptableSensorBase::AllocateInterrupts(bool watcher) {
./wpilibc/src/main/native/cpp/InterruptableSensorBase.cpp:79:void InterruptableSensorBase::CancelInterrupts() {
./wpilibc/src/main/native/cpp/InterruptableSensorBase.cpp:100:InterruptableSensorBase::WaitResult InterruptableSensorBase::WaitForInterrupt(
./wpilibc/src/main/native/cpp/InterruptableSensorBase.cpp:102:  if (StatusIsFatal()) return InterruptableSensorBase::kTimeout;
./wpilibc/src/main/native/cpp/InterruptableSensorBase.cpp:126:void InterruptableSensorBase::EnableInterrupts() {
./wpilibc/src/main/native/cpp/InterruptableSensorBase.cpp:137:void InterruptableSensorBase::DisableInterrupts() {
./wpilibc/src/main/native/cpp/InterruptableSensorBase.cpp:154:double InterruptableSensorBase::ReadRisingTimestamp() {
./wpilibc/src/main/native/cpp/InterruptableSensorBase.cpp:172:double InterruptableSensorBase::ReadFallingTimestamp() {
./wpilibc/src/main/native/cpp/InterruptableSensorBase.cpp:187:void InterruptableSensorBase::SetUpSourceEdge(bool risingEdge,
./wpilibc/src/main/native/include/DigitalSource.h:12:#include "InterruptableSensorBase.h"
./wpilibc/src/main/native/include/DigitalSource.h:25:class DigitalSource : public InterruptableSensorBase {
./wpilibc/src/main/native/include/InterruptableSensorBase.h:17:class InterruptableSensorBase : public SensorBase {
./wpilibc/src/main/native/include/InterruptableSensorBase.h:26:  InterruptableSensorBase() = default;
./wpilibc/src/main/native/include/InterruptableSensorBase.h:27:  virtual ~InterruptableSensorBase() = default;
./wpilibc/src/main/native/include/WPILib.h:54:#include "InterruptableSensorBase.h"
./wpilibjIntegrationTests/src/main/java/edu/wpi/first/wpilibj/AbstractInterruptTest.java:28: * use the InterruptableSensorBase.
./wpilibjIntegrationTests/src/main/java/edu/wpi/first/wpilibj/AbstractInterruptTest.java:31:  private InterruptableSensorBase m_interruptable = null;
./wpilibjIntegrationTests/src/main/java/edu/wpi/first/wpilibj/AbstractInterruptTest.java:33:  private InterruptableSensorBase getInterruptable() {
./wpilibjIntegrationTests/src/main/java/edu/wpi/first/wpilibj/AbstractInterruptTest.java:35:      m_interruptable = giveInterruptableSensorBase();
./wpilibjIntegrationTests/src/main/java/edu/wpi/first/wpilibj/AbstractInterruptTest.java:44:      freeInterruptableSensorBase();
./wpilibjIntegrationTests/src/main/java/edu/wpi/first/wpilibj/AbstractInterruptTest.java:52:  abstract InterruptableSensorBase giveInterruptableSensorBase();
[... cut ...]
./wpilibjIntegrationTests/src/main/java/edu/wpi/first/wpilibj/DIOCrossConnectTest.java:157:   * edu.wpi.first.wpilibj.AbstractInterruptTest#giveInterruptableSensorBase()
./wpilibjIntegrationTests/src/main/java/edu/wpi/first/wpilibj/DIOCrossConnectTest.java:160:  InterruptableSensorBase giveInterruptableSensorBase() {
./wpilibjIntegrationTests/src/main/java/edu/wpi/first/wpilibj/DIOCrossConnectTest.java:168:   * edu.wpi.first.wpilibj.AbstractInterruptTest#freeInterruptableSensorBase()
./wpilibjIntegrationTests/src/main/java/edu/wpi/first/wpilibj/DIOCrossConnectTest.java:171:  void freeInterruptableSensorBase() {
```

Starlight220, 2020-04-19:
If fixed - issue should be closed. 
If stale and not planning on being fixed - issue should be closed.
If stale and needs someone to fix - I can do it, just tell me.

PR #2410 (2021-06-05): Remove InterruptableSensorBase and replace with interrupt classes
Needs Java and formatting, but much cleaner from a user standpoint.

Closes #738
closed 2021-06-05 by PeterJohnson