wpilibsuite/allwpilib #773: Connect the repo to Reviewable

timscharfenort, 2017-11-25:
Please connect this repo at https://reviewable.io/repositories so that all PRs get linked to reviews. Thanks!

Twinters007, 2017-11-25:
> Can I mix using GitHub and Reviewable on a PR?

> Yes, but the experience won't be optimal. While comments posted in one will also appear in the other, line comments will turn into top-level comments due to the systems' irreconcilable approaches to line numbering. Also, Reviewable expects that all participants are using its system for tracking unreviewed files and unresolved comments, so if some people are using GitHub those counters probably won't be right.

Doesn't seem like a great idea...

PeterJohnson, 2017-11-25:
We tried Reviewable early on, but ended up deciding just normal GitHub reviews met our needs.  Thanks for the suggestion, but at the present time, we're seeing no need to use it.

closed 2017-11-25 by PeterJohnson