wpilibsuite/allwpilib #777: Resetting HAL handles doesn't work in Linux

pjreiniger, 2017-11-27:
If you try to unit test your code you would ideally like to "shutdown" the robot and start with a fresh copy between each test.  I've noticed (and ignored for quite some time) that the mechanism for resetting the static features in the HAL does not work on linux, but it does when compiled for windows.  

Specificially, certain modules that I've made unit tests for (PWM, DIO, PCM/Solenoid, AnalogIn) do not reset their handles correctly, which results in you not being able to do:

```
// psedo code
ASSERT_TRUE(HAL_InitializePWMPort(1))
hal::HandleBase::ResetGlobalHandles();
ASSERT_TRUE(HAL_InitializePWMPort(1))
```

I made a branch that adds some unit tests to the HALSIM project here [here](https://github.com/pjreiniger/allwpilib/tree/halsim_unit_tests), and made another branch with some (https://github.com/pjreiniger/allwpilib/tree/halsim_unit_tests_with_debug)[debugging couts], the output of which is below.  I tried turning optimization off to see if it was messing things up on linux, but that seemed to result in the same issue

```
> Task :hal:runHalSimTestingBaseTestGoogleTestExe
Initializing handle base...  --- Adding new... 0x7f0daf14f900, 1
Initializing handle base...  --- Adding new... 0x7f0daf150260, 2
Initializing handle base...  --- Adding new... 0x7f0daf150460, 3
Initializing handle base...  --- Adding new... 0x7f0daf1513c0, 4
Initializing handle base...  --- Adding new... 0x7f0daf1515e0, 5
Initializing handle base...  --- Adding new... 0x7f0daf151660, 6
Initializing handle base...  --- Adding new... 0x7f0daf15a580, 7
Initializing handle base...  --- Adding new... 0x7f0daf162520, 8
<<<< I'd assume this is the root cause, don't know why the global object seems to have two copies >>>>
Initializing handle base...  --- Adding new... 0x7f0daf1630c0, 1
Initializing handle base...  --- Adding new... 0x7f0daf1634c0, 2
Initializing handle base...  --- Adding new... 0x7f0daf163800, 3
Initializing handle base...  --- Adding new... 0x7f0daf16b7e0, 4
Initializing handle base...  --- Adding new... 0x7f0daf16bba0, 5
Initializing handle base...  --- Adding new... 0x7f0daf16bec0, 6
[==========] Running 13 tests from 12 test cases.
[----------] Global test environment set-up.
[----------] 1 test from HALTests
[ RUN      ] HALTests.RuntimeType
[       OK ] HALTests.RuntimeType (0 ms)
[----------] 1 test from HALTests (0 ms total)

[----------] 1 test from HALSOLENOID
[ RUN      ] HALSOLENOID.SolenoidTest
[       OK ] HALSOLENOID.SolenoidTest (0 ms)
[----------] 1 test from HALSOLENOID (0 ms total)

[----------] 1 test from AnalogOutSimTests
[ RUN      ] AnalogOutSimTests.TestAnalogOutInitialization
<<<< This only resets the second "copy" of the global vector.  Again, likely the root cause >>>>
Resetting global handles... 6
Resetting IndexedHandle handles... 2, 5, 0x7f0daf1630c0
limited handle digital handles... 8, 0x7f0daf1634c0
limited handle digital handles... 8, 0x7f0daf163800
limited handle digital handles... 16, 0x7f0daf16b7e0
limited handle digital handles... 8, 0x7f0daf16bba0
Resetting IndexedHandle handles... 8, 8, 0x7f0daf16bec0
[       OK ] AnalogOutSimTests.TestAnalogOutInitialization (1 ms)
[----------] 1 test from AnalogOutSimTests (1 ms total)

[----------] 1 test from PWMSimTests
[ RUN      ] PWMSimTests.TestPwmInitialization
Resetting global handles... 6
Resetting IndexedHandle handles... 2, 5, 0x7f0daf1630c0
limited handle digital handles... 8, 0x7f0daf1634c0
limited handle digital handles... 8, 0x7f0daf163800
limited handle digital handles... 16, 0x7f0daf16b7e0
limited handle digital handles... 8, 0x7f0daf16bba0
Resetting IndexedHandle handles... 8, 8, 0x7f0daf16bec0
/home/pj/git/allwpilib/hal/src/test/native/cpp/MockData/PWMDataTests.cpp:75: Failure
Value of: status
  Actual: -1029
Expected: 0
/home/pj/git/allwpilib/hal/src/test/native/cpp/MockData/PWMDataTests.cpp:76: Failure
Value of: gTestPwmCallbackName.c_str()
[... 106 lines cut ...]
[==========] 13 tests from 12 test cases ran. (3 ms total)
[  PASSED  ] 9 tests.
[  FAILED  ] 4 tests, listed below:
[  FAILED  ] PWMSimTests.TestPwmInitialization
[  FAILED  ] DigitalIoSimTests.TestDigitalIoInitialization
[  FAILED  ] SolenoidSimTests.TestSolenoidInitialization
[  FAILED  ] AnalogInSimTests.TestAnalogInInitialization

 4 FAILED TESTS
shuting down global handle 0x7f0daf16bec0, 6
shuting down global handle 0x7f0daf16bba0, 6
shuting down global handle 0x7f0daf16b7e0, 6
shuting down global handle 0x7f0daf163800, 6
shuting down global handle 0x7f0daf1634c0, 6
shuting down global handle 0x7f0daf1630c0, 6
shuting down global handle 0x7f0daf162520, 6
shuting down global handle 0x7f0daf15a580, 6
shuting down global handle 0x7f0daf151660, 6
shuting down global handle 0x7f0daf1515e0, 6
shuting down global handle 0x7f0daf1513c0, 6
shuting down global handle 0x7f0daf150460, 6
shuting down global handle 0x7f0daf150260, 6
shuting down global handle 0x7f0daf14f900, 6

```

pjreiniger, 2017-11-27:
Windows debugging.  Note the 6+8 size difference

```
> Task :hal:runHalSimTestingBaseTestGoogleTestExe
Initializing handle base...  --- Adding new... 00007FFCD3C25920, 1
Initializing handle base...  --- Adding new... 00007FFCD3C25D60, 2
Initializing handle base...  --- Adding new... 00007FFCD3C286F0, 3
Initializing handle base...  --- Adding new... 00007FFCD3C28FF0, 4
Initializing handle base...  --- Adding new... 00007FFCD3C29350, 5
Initializing handle base...  --- Adding new... 00007FFCD3C35600, 6
Initializing handle base...  --- Adding new... 00007FFCD3C36490, 7
Initializing handle base...  --- Adding new... 00007FFCD3C36590, 8
Initializing handle base...  --- Adding new... 00007FFCD3C36BF0, 9
Initializing handle base...  --- Adding new... 00007FFCD3C37360, 10
Initializing handle base...  --- Adding new... 00007FFCD3C45820, 11
Initializing handle base...  --- Adding new... 00007FFCD3C46320, 12
Initializing handle base...  --- Adding new... 00007FFCD3C46420, 13
Initializing handle base...  --- Adding new... 00007FFCD3C467D0, 14
[==========] Running 13 tests from 12 test cases.
[----------] Global test environment set-up.
[----------] 1 test from I2CSimTests
[ RUN      ] I2CSimTests.TestI2CInitialization
[       OK ] I2CSimTests.TestI2CInitialization (0 ms)
[----------] 1 test from I2CSimTests (0 ms total)

[----------] 1 test from PdpSimTests
[ RUN      ] PdpSimTests.TestPdpInitialization
[       OK ] PdpSimTests.TestPdpInitialization (0 ms)
[----------] 1 test from PdpSimTests (0 ms total)

[----------] 1 test from SolenoidSimTests
[ RUN      ] SolenoidSimTests.TestSolenoidInitialization
Resetting global handles... 14
Resetting IndexedHandle handles... 8, 6, 00007FFCD3C25920
Resetting digital handles... 36, 00007FFCD3C25D60
unlimmted handles... 0, 00007FFCD3C286F0
limited handle digital handles... 8, 00007FFCD3C28FF0
unlimmted handles... 0, 00007FFCD3C29350
limited handle digital handles... 8, 00007FFCD3C35600
Resetting IndexedHandle handles... 2, 16, 00007FFCD3C36490
limited handle digital handles... 16, 00007FFCD3C36590
limited handle digital handles... 8, 00007FFCD3C36BF0
Resetting IndexedHandle handles... 504, 15, 00007FFCD3C37360
limited handle digital handles... 8, 00007FFCD3C45820
Resetting IndexedHandle handles... 2, 5, 00007FFCD3C46320
Resetting IndexedHandle handles... 8, 8, 00007FFCD3C46420
limited handle digital handles... 6, 00007FFCD3C467D0
Second initialize...
[       OK ] SolenoidSimTests.TestSolenoidInitialization (2 ms)
[----------] 1 test from SolenoidSimTests (2 ms total)

[----------] 1 test from AnalogOutSimTests
[ RUN      ] AnalogOutSimTests.TestAnalogOutInitialization
Resetting global handles... 14
Resetting IndexedHandle handles... 8, 6, 00007FFCD3C25920
Resetting digital handles... 36, 00007FFCD3C25D60
unlimmted handles... 0, 00007FFCD3C286F0
limited handle digital handles... 8, 00007FFCD3C28FF0
unlimmted handles... 0, 00007FFCD3C29350
limited handle digital handles... 8, 00007FFCD3C35600
Resetting IndexedHandle handles... 2, 16, 00007FFCD3C36490
limited handle digital handles... 16, 00007FFCD3C36590
limited handle digital handles... 8, 00007FFCD3C36BF0
Resetting IndexedHandle handles... 504, 15, 00007FFCD3C37360
limited handle digital handles... 8, 00007FFCD3C45820
Resetting IndexedHandle handles... 2, 5, 00007FFCD3C46320
Resetting IndexedHandle handles... 8, 8, 00007FFCD3C46420
limited handle digital handles... 6, 00007FFCD3C467D0
[       OK ] AnalogOutSimTests.TestAnalogOutInitialization (2 ms)
[----------] 1 test from AnalogOutSimTests (2 ms total)
[... 96 lines cut ...]
limited handle digital handles... 8, 00007FFCD3C45820
Resetting IndexedHandle handles... 2, 5, 00007FFCD3C46320
Resetting IndexedHandle handles... 8, 8, 00007FFCD3C46420
limited handle digital handles... 6, 00007FFCD3C467D0
[       OK ] PWMSimTests.TestPwmInitialization (1 ms)
[----------] 1 test from PWMSimTests (1 ms total)

[----------] Global test environment tear-down
[==========] 13 tests from 12 test cases ran. (10 ms total)
[  PASSED  ] 13 tests.
shuting down global handle 00007FFCD3C467D0, 14
shuting down global handle 00007FFCD3C46420, 14
shuting down global handle 00007FFCD3C46320, 14
shuting down global handle 00007FFCD3C45820, 14
shuting down global handle 00007FFCD3C37360, 14
shuting down global handle 00007FFCD3C36BF0, 14
shuting down global handle 00007FFCD3C36590, 14
shuting down global handle 00007FFCD3C36490, 14
shuting down global handle 00007FFCD3C35600, 14
shuting down global handle 00007FFCD3C29350, 14
shuting down global handle 00007FFCD3C28FF0, 14
shuting down global handle 00007FFCD3C286F0, 14
shuting down global handle 00007FFCD3C25D60, 14
shuting down global handle 00007FFCD3C25920, 14
```

ThadHouse, 2017-11-27:
Will you double check to make sure your use case isn't statically linking the HAL anywhere? That could potentially cause something like this. I'll double check the WPILib build to make sure we are not anywhere.

pjreiniger, 2017-11-27:
The unit test are in the HAL project, so it would just be however that guys unit test gets linked. Looks like it calls addHalToLinker, and does a 'shared' linkage

pjreiniger, 2017-11-27:
On that statically linked not though, the wpilibj definitely does a static linkage, so I have trouble talking to the HAL libraries from my simulator.  For now I've been locally changing it to shared and using local libraries for my simulator.  I was going to bring that up at some point, but I've had plenty of other things to work on and its a quick and easy fix/hack for me to do.

ThadHouse, 2017-11-27:
There's 2 versions of the jni library. You want to be using jnishared of you need shared access. Jnistatic only has specific use cases, which shouldn't be getting hit by the test code. If it is, we should change that.

ThadHouse, 2017-11-27:
So I just got a chance to run the code in WSL (I don't have a full linux machine). I'm seeing it initialize all 10 without a double initialization, but I do get some tests failing in the same way as yours. The relay tests are failing, which were not on yours. I'll keep looking.

```
Initializing handle base...  --- Adding new... 0x7f9fa01e2900, 1
Initializing handle base...  --- Adding new... 0x7f9fa01eaca0, 2
Initializing handle base...  --- Adding new... 0x7f9fa01ecda0, 3
Initializing handle base...  --- Adding new... 0x7f9fa01ed160, 4
Initializing handle base...  --- Adding new... 0x7f9fa01edc40, 1
Initializing handle base...  --- Adding new... 0x7f9fa01ede40, 2
Initializing handle base...  --- Adding new... 0x7f9fa01edec0, 3
Initializing handle base...  --- Adding new... 0x7f9fa01f6680, 4
Initializing handle base...  --- Adding new... 0x7f9fa01f6840, 5
Initializing handle base...  --- Adding new... 0x7f9fa01f68a0, 6
Initializing handle base...  --- Adding new... 0x7f9fa01f6aa0, 7
Initializing handle base...  --- Adding new... 0x7f9fa01f7180, 8
Initializing handle base...  --- Adding new... 0x7f9fa01feba0, 9
Initializing handle base...  --- Adding new... 0x7f9fa01fedc0, 10
[==========] Running 13 tests from 12 test cases.
[----------] Global test environment set-up.
[----------] 1 test from HALTests
[ RUN      ] HALTests.RuntimeType
[       OK ] HALTests.RuntimeType (0 ms)
[----------] 1 test from HALTests (0 ms total)

[----------] 1 test from HALSOLENOID
[ RUN      ] HALSOLENOID.SolenoidTest
[       OK ] HALSOLENOID.SolenoidTest (0 ms)
[----------] 1 test from HALSOLENOID (0 ms total)

[----------] 1 test from I2CSimTests
[ RUN      ] I2CSimTests.TestI2CInitialization
[       OK ] I2CSimTests.TestI2CInitialization (0 ms)
[----------] 1 test from I2CSimTests (0 ms total)

[----------] 1 test from RelaySimTests
[ RUN      ] RelaySimTests.TestRelayInitialization
Resetting global handles... 10
limited handle digital handles... 8, 0x7f9fa01edc40
unlimmted handles... 0, 0x7f9fa01ede40
Resetting digital handles... 36, 0x7f9fa01edec0
limited handle digital handles... 6, 0x7f9fa01f6680
unlimmted handles... 0, 0x7f9fa01f6840
Resetting IndexedHandle handles... 8, 6, 0x7f9fa01f68a0
Resetting IndexedHandle handles... 2, 5, 0x7f9fa01f6aa0
limited handle digital handles... 8, 0x7f9fa01f7180
limited handle digital handles... 8, 0x7f9fa01feba0
Resetting IndexedHandle handles... 2, 16, 0x7f9fa01fedc0
/home/thad/git/allwpilib/hal/src/test/native/cpp/MockData/RelayDataTests.cpp:75: Failure
Value of: status
  Actual: -1029
Expected: 0
/home/thad/git/allwpilib/hal/src/test/native/cpp/MockData/RelayDataTests.cpp:76: Failure
Value of: gTestRelayCallbackName.c_str()
  Actual: "Unset"
Expected: "InitializedForward"
[  FAILED  ] RelaySimTests.TestRelayInitialization (0 ms)
[----------] 1 test from RelaySimTests (0 ms total)

[----------] 1 test from SolenoidSimTests
[ RUN      ] SolenoidSimTests.TestSolenoidInitialization
Resetting global handles... 10
limited handle digital handles... 8, 0x7f9fa01edc40
unlimmted handles... 0, 0x7f9fa01ede40
Resetting digital handles... 36, 0x7f9fa01edec0
limited handle digital handles... 6, 0x7f9fa01f6680
unlimmted handles... 0, 0x7f9fa01f6840
Resetting IndexedHandle handles... 8, 6, 0x7f9fa01f68a0
Resetting IndexedHandle handles... 2, 5, 0x7f9fa01f6aa0
limited handle digital handles... 8, 0x7f9fa01f7180
limited handle digital handles... 8, 0x7f9fa01feba0
[... 97 lines cut ...]
[----------] 1 test from PdpSimTests (0 ms total)

[----------] Global test environment tear-down
[==========] 13 tests from 12 test cases ran. (1 ms total)
[  PASSED  ] 11 tests.
[  FAILED  ] 2 tests, listed below:
[  FAILED  ] RelaySimTests.TestRelayInitialization
[  FAILED  ] SolenoidSimTests.TestSolenoidInitialization

 2 FAILED TESTS
shuting down global handle 0x7f9fa01fedc0, 10
shuting down global handle 0x7f9fa01feba0, 10
shuting down global handle 0x7f9fa01f7180, 10
shuting down global handle 0x7f9fa01f6aa0, 10
shuting down global handle 0x7f9fa01f68a0, 10
shuting down global handle 0x7f9fa01f6840, 10
shuting down global handle 0x7f9fa01f6680, 10
shuting down global handle 0x7f9fa01edec0, 10
shuting down global handle 0x7f9fa01ede40, 10
shuting down global handle 0x7f9fa01edc40, 10
shuting down global handle 0x7f9fa01ed160, 10
shuting down global handle 0x7f9fa01ecda0, 10
shuting down global handle 0x7f9fa01eaca0, 10
shuting down global handle 0x7f9fa01e2900, 10
```

ThadHouse, 2017-11-27:
One thing to try is we've had issues with statics across translation units before. Can you try replacing HandlesInternal.cpp with the following code and seeing what happens?

```
/*----------------------------------------------------------------------------*/
/* Copyright (c) 2016-2017 FIRST. All Rights Reserved.                        */
/* Open Source Software - may be modified and shared by FRC teams. The code   */
/* must be accompanied by the FIRST BSD license file in the root directory of */
/* the project.                                                               */
/*----------------------------------------------------------------------------*/

#include "HAL/handles/HandlesInternal.h"

#include <algorithm>

#include <llvm/SmallVector.h>
#include <support/mutex.h>

namespace hal {
// static llvm::SmallVector<HandleBase*, 32> globalHandles;

static llvm::SmallVector<HandleBase*, 32>& GetGlobalHandles() {
  static llvm::SmallVector<HandleBase*, 32> globalHandles;
  return globalHandles;
}

static wpi::mutex& GetGlobalHandleMutex() {
  static wpi::mutex globalHandleMutex;
  return globalHandleMutex;
}

HandleBase::HandleBase() {
  std::lock_guard<wpi::mutex> lock(GetGlobalHandleMutex());
  auto index =
      std::find(GetGlobalHandles().begin(), GetGlobalHandles().end(), this);
  if (index == GetGlobalHandles().end()) {
    GetGlobalHandles().push_back(this);
  } else {
    *index = this;
  }
}

HandleBase::~HandleBase() {
  std::lock_guard<wpi::mutex> lock(GetGlobalHandleMutex());
  auto index =
      std::find(GetGlobalHandles().begin(), GetGlobalHandles().end(), this);
  if (index != GetGlobalHandles().end()) {
    *index = nullptr;
  }
}

void HandleBase::ResetHandles() {
  m_version++;
  if (m_version > 255) {
    m_version = 0;
  }
}

void HandleBase::ResetGlobalHandles() {
  std::unique_lock<wpi::mutex> lock(GetGlobalHandleMutex());
  for (auto&& i : GetGlobalHandles()) {
    if (i != nullptr) {
      lock.unlock();
      i->ResetHandles();
      lock.lock();
    }
  }
}

HAL_PortHandle createPortHandle(uint8_t channel, uint8_t module) {
  // set last 8 bits, then shift to first 8 bits
  HAL_PortHandle handle = static_cast<HAL_PortHandle>(HAL_HandleEnum::Port);
  handle = handle << 24;
  // shift module and add to 3rd set of 8 bits
  int32_t temp = module;
  temp = (temp << 8) & 0xff00;
  handle += temp;
  // add channel to last 8 bits
  handle += channel;
  return handle;
}

HAL_PortHandle createPortHandleForSPI(uint8_t channel) {
  // set last 8 bits, then shift to first 8 bits
  HAL_PortHandle handle = static_cast<HAL_PortHandle>(HAL_HandleEnum::Port);
  handle = handle << 16;
  // set second set up bits to 1
  int32_t temp = 1;
  temp = (temp << 8) & 0xff00;
  handle += temp;
  // shift to last set of bits
  handle = handle << 8;
  // add channel to last 8 bits
  handle += channel;
  return handle;
}

HAL_Handle createHandle(int16_t index, HAL_HandleEnum handleType,
                        int16_t version) {
  if (index < 0) return HAL_kInvalidHandle;
  uint8_t hType = static_cast<uint8_t>(handleType);
  if (hType == 0 || hType > 127) return HAL_kInvalidHandle;
  // set last 8 bits, then shift to first 8 bits
  HAL_Handle handle = hType;
  handle = handle << 8;
  handle += static_cast<uint8_t>(version);
  handle = handle << 16;
  // add index to set last 16 bits
  handle += index;
  return handle;
}
}  // namespace hal
```

pjreiniger, 2017-11-27:
I think that did it.  I'll clean some things and do a PR tomorrow.

Thanks for the help @ThadHouse

PR #779 (2017-12-03): Fixes cross module base class statics
I still don't understand why exactly this has mostly affected this file. It hasn't affected any of our other handle classes,  but we've had issues here with both the mutex and now the actual handle cache.

Closes #777
closed 2017-12-03 by PeterJohnson