wpilibsuite/allwpilib #787: Implement XboxControllerButton

calcmogul, 2017-11-29:
A class exists for registering joystick button rising and trailing edge events, JoystickButton, but it's hard to use with an Xbox controller due to the controller's button mapping. A wrapper around Button should be written with a similar interface to XboxController to perform the button mapping internally for the user.

karagenit, 2017-12-01:
Do you have some documentation that shows how the XboxController button map is different from a typical Joystick? And do you know if this differs between Xbox 360 and Xbox One controllers?

calcmogul, 2017-12-01:
This would have a similar interface to XboxController (see https://github.com/wpilibsuite/allwpilib/blob/master/wpilibc/src/main/native/cpp/XboxController.h) and use the mapping in https://github.com/wpilibsuite/allwpilib/blob/master/wpilibc/src/main/native/cpp/XboxController.cpp. The mapping is for Xbox 360 controllers.  I'm not sure about Xbox One, but I'd imagine it's similar.

karagenit, 2017-12-03:
Okay, looking at things like Joystick.java (the C++ version has already completely removed it) it seems like we're moving away from public enums (e.g. `getButton(ButtonType button)`) to having multiple methods for each possible value in the enum (e.g. `getTop()` & `getTrigger()`). 

My question is: why? It seems like it would be a much simpler API to have just a single method to get the all buttons (after all, that's what GenericHID does with `getRawButton()`)... I'm sure there was a good cause for it, I'd just like to know the reasoning.

Anyway, I'm assuming that for similar reasons we don't want to make the XboxController.Button enum public? If we could, it would make implementing this much simpler. 

Right now, the only ways I can think of doing it are:

**Static "Factory" Style Methods"**  
- Static construction methods (e.g. `XboxControllerButton.getA(joystick)`) which call the constructor with an enum/value
- A separate Button enum to XboxControllerButton which mirrors XboxController's enum
- A big switch/case in `get()` for each enum value, calling the corresponding XboxController method.

Note that even making the XboxController.Button enum `protected` wouldn't allow XboxControllerButton to access it, because XboxControllerButton must inherit from Button and not XboxController.

**XboxControllerButtonMap interface**  
To allow us to access the XboxController.Button enum from XboxControllerButton, we could create an interface which contains a protected enum (still not publicly accessible) which is then implemented by both XboxController and XboxControllerButton. This makes me feel "safer" knowing that one of the enums couldn't get changed to make it not match the other, but I think it's still a pretty ugly/hacky solution.

**Make Enum Package Private, Move XboxControllerButton Out Of Buttons Sub-Package**  
This is also kind of ugly, because it means XboxControllerButton won't be grouped with all the other Buttons and such, but it would require the least amount of code changed AND prevent the Button enum from being public AND make sure the enums always match up. Not sure how this would play into our C++ design, as I don't think it has any equivalent of package-private (something something friend classes?).

calcmogul, 2017-12-03:
Both Joystick.h and Joystick.java have only deprecated the getButton() interfaces, not removed them. XboxController.java and XboxController.h never had them.

I originally wanted to provide `getButton()` with an enum argument because it's simpler from a developer implementation perspective, but it's more verbose from a user perspective. For example, `controller.getButton(XboxController.Button.kA);` with an enum versus `controller.getAButton();`. `getButtonPressed()` and `getButtonReleased()` are even longer.

> making the XboxController.Button enum public would make this even more of a non-issue - users could simply call JoystickButton() with Button.A.value
> we could make another constructor in JoystickButton that takes an XboxController.Button instead of an int for the buttonNumber.

An XboxController isn't a Joystick, so those interfaces shouldn't be mixed.

Of all these options, providing an enum in XboxControllerButton seems the most palatable. The conversion from enum to int should happen in the constructor, not `Get()`.

PeterJohnson, 2019-08-26:
The enum was made public in #1823, so this is no longer necessary.

closed 2019-08-26 by PeterJohnson