wpilibsuite/allwpilib #843: SendableChooser change listener no longer (easily) exists

pjreiniger, 2017-12-19:
Last year I was able to do something along the lines of 

```
SendableChooser chooser = new SendableChooser()
// ... fill in options ....
SmartDashboard.putData("Chooser Name", chooser );

ITableListener changeListener= ....
chooser.getTable().addTableListener(changeListener);
```

Looks like the functionality was removed in #622.  I can certainly modify our side to look more like this

```
NetworkTableInstance.getDefault().getTable("SmartDashboard").getSubTable("ChooserName").addEntryListener("selected", changeListener, 0xFF);
```
But that seems clunky, especially with two magic strings in there ("SmartDashboard" is not a constant, "selected" is a private constant)

I'm not sure how large this use case is, but the convenience of having `chooser.addSelectionChangedListener(listener, flags)` might be nice to have.  I think I understand the path to add this and don't mind implementing it, but I wanted to double check that this doesn't exist somewhere or was removed on purpose

ThadHouse, 2017-12-19:
That definitely seems like an oversight. I'm fine with that as a PR, and think it would be a valuable addition. Not sure it would get merged before the season but will definitely look at it

PeterJohnson, 2017-12-19:
Unless... the listener is just a Runnable and we set up a thunk callback in initSendable().  Then you could make it work for either order.

PeterJohnson, 2017-12-19:
Also, not sure why you need flags.  Don't you only care about valueChanged?

Bankst, 2022-03-21:
I suggest an `onChange(Consumer<T> consumer)` method to expose this

PeterJohnson, 2023-07-18:
Fixed by #5458.

closed 2023-07-18 by PeterJohnson