wpilibsuite/allwpilib #847: Failed to load plugins

mayaankvad, 2017-12-23:
I have gazebo and frcsim installed on ubuntu. Eclipse allows me to run the simulation fine. The issue is that when autonomous is enabled nothing happens. 

The frcsim terminal has many errors like this:

`
Failed to load plugin /home/mayaankvad/wpilib/simulation/plugins/libclock.so: /home/mayaankvad/Downloads/allwpilib/build/simulation/gz_msgs/libgz_msgs.so: undefined symbol: _ZNK6google8protobuf7Message11GetTypeNameEv
`

However all the files are inside the `wpilib/simulation/plugins/` directory.

Also The terminal has this (not sure if it is relevant): 

`
(gzclient:2833): Gtk-WARNING **: Unable to locate theme engine in module_path: "adwaita",
`

How do I resolve these errors and get the robot moving? For now I have only tested it with the default PacGoat robot

coder3462, 2017-12-23:
I have the same issue.

calculus-lumberjack, 2017-12-23:
I also have this error

Atharva-pandhare, 2017-12-23:
I am having this error too.

Atharva-pandhare, 2017-12-23:
I have this error too.

AustinShalit, 2019-08-04:
Duplicate of #456

closed 2019-08-04 by AustinShalit