wpilibsuite/allwpilib #862: Simulation v2017.3.1 Compile issue

lalbrant, 2018-01-01:
Using latest allwpilib (as of 1/1/18), performed the following steps:

1) git clone https://github.com/wpilibsuite/allwpilib
2) git checkout v2017.3.1
3) ./gradlew build -PmakeSIM
4) Receive the following errors:
/home/lalbrant/Downloads/allwpilib/wpilibc/build/libs/wpilibc/shared/libwpilibc.so: undefined reference to `cs::GetSinkName[abi:cxx11](int, int*)'
/home/lalbrant/Downloads/allwpilib/wpilibc/build/libs/wpilibc/shared/libwpilibc.so: undefined reference to `cs::GetSinkError[abi:cxx11](int, int*)'
/home/lalbrant/Downloads/allwpilib/wpilibc/build/libs/wpilibc/shared/libwpilibc.so: undefined reference to `cs::GetSourceName[abi:cxx11](int, int*)'
/home/lalbrant/Downloads/allwpilib/wpilibc/build/libs/wpilibc/shared/libwpilibc.so: undefined reference to `cs::GetStringProperty[abi:cxx11](int, int*)'
/home/lalbrant/Downloads/allwpilib/wpilibc/build/libs/wpilibc/shared/libwpilibc.so: undefined reference to `cs::GetUsbCameraPath[abi:cxx11](int, int*)'
/home/lalbrant/Downloads/allwpilib/wpilibc/build/libs/wpilibc/shared/libwpilibc.so: undefined reference to `cs::GetNetworkInterfaces[abi:cxx11]()'
/home/lalbrant/Downloads/allwpilib/wpilibc/build/libs/wpilibc/shared/libwpilibc.so: undefined reference to `cs::CreateHttpCamera(llvm::StringRef, llvm::ArrayRef<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, CS_HttpCameraKind, int*)'
/home/lalbrant/Downloads/allwpilib/wpilibc/build/libs/wpilibc/shared/libwpilibc.so: undefined reference to `nt::Value::MakeStringArray(llvm::ArrayRef<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >)'
/home/lalbrant/Downloads/allwpilib/wpilibc/build/libs/wpilibc/shared/libwpilibc.so: undefined reference to `cs::GetHttpCameraUrls[abi:cxx11](int, int*)'
/home/lalbrant/Downloads/allwpilib/wpilibc/build/libs/wpilibc/shared/libwpilibc.so: undefined reference to `cs::GetHostname[abi:cxx11]()'
/home/lalbrant/Downloads/allwpilib/wpilibc/build/libs/wpilibc/shared/libwpilibc.so: undefined reference to `cs::GetEnumPropertyChoices[abi:cxx11](int, int*)'
/home/lalbrant/Downloads/allwpilib/wpilibc/build/libs/wpilibc/shared/libwpilibc.so: undefined reference to `cs::GetMjpegServerListenAddress[abi:cxx11](int, int*)'
collect2: error: ld returned 1 exit status

:myRobotCpp:linkMyRobotcppExecutable FAILED

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':myRobotCpp:linkMyRobotcppExecutable'.
> A build operation failed.
      Linker failed while linking myRobotcpp.
  See the complete log at: file:///home/lalbrant/Downloads/allwpilib/myRobotCpp/build/tmp/linkMyRobotcppExecutable/output.txt

PeterJohnson, 2018-01-08:
You need to update your arm compiler packages.  The frc-toolchain was updated from gcc 4.9 to gcc 5.4 for 2018, which changed the C++ ABI.

lalbrant, 2018-01-09:
gcc --version
gcc (Ubuntu 7.2.0-8ubuntu3) 7.2.0

Safe to assume I need to step it down to 5.4 from 7.2?

PeterJohnson, 2018-01-10:
No, I'm talking about `arm-frc-linux-gnueabi-gcc` (the compiler for the roborio), which is part of the frc-toolchain package.  Note: as of this writing, you'll need to use the frc-toolchain beta repo to get updated packages (see #882).

lalbrant, 2018-01-10:
Thank you Peter.

Changed to the frc-toolchain-beta repo as recommended and performed apt update and upgrade, now have version 5.5.0 as seen below:

lalbrant@lalbrant-GS60-6QE:~/Downloads/allwpilib$ arm-frc-linux-gnueabi-gcc --version
arm-frc-linux-gnueabi-gcc (GCC for FRC) 5.5.0
----------------------------------------------------------------------------------------------------------------------
Ran the checkout process:

lalbrant@lalbrant-GS60-6QE:~/Downloads/allwpilib$ git checkout v2017.3.1
Note: checking out 'v2017.3.1'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b <new-branch-name>

HEAD is now at 12f75986... Corrects assumptions about return values from i2c-lib (#484)

----------------------------------------------------------------------------------------------------------------------

Ran the build again with the following output:

lalbrant@lalbrant-GS60-6QE:~/Downloads/allwpilib$ ./gradlew build -PmakeSim
Starting a Gradle Daemon, 1 incompatible and 1 stopped Daemons could not be reused, use --status for details
:downloadArmWpiutil
:downloadNativeWpiutil
:downloadWpiutil
:hal:compileHALAthenaSharedLibraryHALAthenaCpp
:hal:linkHALAthenaSharedLibrary
:hal:HALAthenaSharedLibrary
:hal:compileHALAthenaStaticLibraryHALAthenaCpp
:hal:createHALAthenaStaticLibrary
:hal:HALAthenaStaticLibrary
:hal:assemble
:downloadArmNetworkTables
:downloadNativeNetworkTables
:downloadNetworkTables
:hal:athenaRuntimeZip
:hal:check UP-TO-DATE
:hal:halZip
:hal:build
:wpilibj:generateJavaVersion
Writing version 424242.2017.3.1-20180110160823 to /home/lalbrant/Downloads/allwpilib/wpilibj/src/shared/java/edu/wpi/first/wpilibj/util/WPILibVersion.java
:wpilibj:compileSharedJava
Note: /home/lalbrant/Downloads/allwpilib/wpilibj/src/shared/java/edu/wpi/first/wpilibj/smartdashboard/SmartDashboard.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: Some input files use unchecked or unsafe operations.
Note: Recompile with -Xlint:unchecked for details.
:wpilibj:processSharedResources UP-TO-DATE
:wpilibj:sharedClasses
:wpilibj:compileAthenaJava
Note: /home/lalbrant/Downloads/allwpilib/wpilibj/src/athena/java/edu/wpi/first/wpilibj/Preferences.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
:wpilibj:processAthenaResources UP-TO-DATE
:wpilibj:athenaClasses
:wpilibj:compileJava UP-TO-DATE
:wpilibj:processResources UP-TO-DATE
:wpilibj:classes UP-TO-DATE
:wpilibj:jar
:myRobot:compileJava
:myRobot:processResources UP-TO-DATE
:myRobot:classes
:wpilibj:jniHeaders
:wpilibj:compileWpilibJavaJNISharedLibraryWpilibJavaJNICpp
:wpilibj:linkWpilibJavaJNISharedLibrary
[... 75 lines cut ...]
/home/lalbrant/Downloads/allwpilib/wpilibc/build/libs/wpilibc/shared/libwpilibc.so: undefined reference to `cs::CreateHttpCamera(llvm::StringRef, llvm::ArrayRef<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, CS_HttpCameraKind, int*)'
/home/lalbrant/Downloads/allwpilib/wpilibc/build/libs/wpilibc/shared/libwpilibc.so: undefined reference to `nt::Value::MakeStringArray(llvm::ArrayRef<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >)'
/home/lalbrant/Downloads/allwpilib/wpilibc/build/libs/wpilibc/shared/libwpilibc.so: undefined reference to `cs::GetHttpCameraUrls[abi:cxx11](int, int*)'
/home/lalbrant/Downloads/allwpilib/wpilibc/build/libs/wpilibc/shared/libwpilibc.so: undefined reference to `cs::GetHostname[abi:cxx11]()'
/home/lalbrant/Downloads/allwpilib/wpilibc/build/libs/wpilibc/shared/libwpilibc.so: undefined reference to `cs::GetEnumPropertyChoices[abi:cxx11](int, int*)'
/home/lalbrant/Downloads/allwpilib/wpilibc/build/libs/wpilibc/shared/libwpilibc.so: undefined reference to `cs::GetMjpegServerListenAddress[abi:cxx11](int, int*)'
collect2: error: ld returned 1 exit status

**:myRobotCpp:linkMyRobotcppExecutable FAILED

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':myRobotCpp:linkMyRobotcppExecutable'.
> A build operation failed.
      Linker failed while linking myRobotcpp.
  See the complete log at: file:///home/lalbrant/Downloads/allwpilib/myRobotCpp/build/tmp/linkMyRobotcppExecutable/output.txt**

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output.

BUILD FAILED

Total time: 1 mins 12.966 secs
lalbrant@lalbrant-GS60-6QE:~/Downloads/allwpilib$

somebuckaroo, 2018-02-13:
The problem you are encountering is described [here](https://gcc.gnu.org/onlinedocs/libstdc++/manual/using_dual_abi.html).

The fix is to add `-D_GLIBCXX_USE_CXX11_ABI=0` to cppCompiler.args in the relevant build.gradle files.

closed 2018-09-28 by PeterJohnson