wpilibsuite/allwpilib #873: Make shadow methods in Command.java and methods in PIDController.java public

AndrewNolte, 2018-01-06:
I want to make useful extensions for these classes, but I always have to come up with workarounds, when it would be so much easier if all of the methods were public. I don't see why they are protected.

calcmogul, 2018-01-06:
Does overriding the currently protected methods via a subclass of PIDController not allow you to add the functionality you want?

AndrewNolte, 2018-01-06:
Oh sorry, that was an error in my wording. The methods are overridable, except if I wanted to change the way it calculated something for example, I would copy and paste the calculate() code into my implementation, and change some things. This doesn't work however, because every variable used in the method isn't visible. I've made a workaround, but it would be nice if everything was public. I'll edit the issue.

calcmogul, 2018-01-07:
Mutable variables don't belong in a public interface. That majorly breaks encapsulation and makes our maintenance job harder. I understand what you're saying with regard to required variables not having the correct access specifier. To make overriding calculate() work again, we can make the variables it uses `protected`.

AndrewNolte, 2018-01-08:
That would be great if they were `protected`! Could you also do that for the shadow methods in `Command.java`? Also, could the method for calculation in PIDController be broken up into different methods?

calcmogul, 2018-01-09:
C++ already made the shadow methods protected, so we can fix Java to reflect that. PIDController probably isn't in a state anymore to allow inheriting from calculate() without teams easily breaking something or causing deadlocks. Perhaps we should provide a function that just does the PID math part that teams can override. In any case, changes to PIDController beyond fixing the access specifiers will have to wait until 2019.

sciencewhiz, 2018-01-09:
Can you explain why you want to modify the shadow methods in Command?

AndrewNolte, 2018-01-09:
I'm making a system where I have triggers and assists that I can add to commands. It would make it easier if I could override the shadow methods to run the triggers and assists, rather than have to call super.initialize(), etc, in the commands using the new class. 
[(https://github.com/Vortx3735/Robot2017/tree/master/Robot2017/src/org/usfirst/frc/team3735/robot/util/cmds)]

You would combine commands with triggers and assists to manufacture the command you need. For example, `addSequential(new DriveRaw(.8,0).addA(new NavxAssist()).addT(new HasMoved(60)).addT(new Bumped(.7)));`

Rather than copying or remaking the code for each command.

calcmogul, 2018-01-09:
If you override a method and need the old behavior to run in addition to the new one, it's convention in Java to call super at the top of the method.

ddyer8, 2018-01-17:
I second the suggestion to make methods to override PID math.   I think the Controller Interface also needs to be public too.  And please consider creating a way for PIDCommand and PIDSubsystem to use the extended PIDController class.  One way would be to make a constructor to pass in a PIDController object instead of building it internally, but I'm sure there are other potentially safer ways.

My current workaround requires copy/pasting the majority of PIDCommand, Controller, PIDSubsystem, and PIDCommand into my own package.