wpilibsuite/allwpilib #880: Undeprecate SampleBot

twinters007, 2018-01-08:
Currently Samplebot is deprecated with a warning message saying that a different robot base is probably better because complex code will be harder. I think a deprecationwarning doesn't convey this message well. Teams should be free to use it, but be warned of its paradigms. Now, deprecationwarnings mean you guys won't update it (and maybe you won't), but I have a feeling that there won't be a lot of changes needed in the future anyway.

PeterJohnson, 2018-01-08:
As you're aware, we had an extensive discussion of this here: https://github.com/wpilibsuite/allwpilib/pull/472

What the deprecation means is that teams are free to use SampleRobot, at least this year, and we will update it as necessary so it works in 2018.  We're looking at removing it in a future year (perhaps 2019, perhaps later), so we want to deprecate it so teams have advance warning of its removal (and can give feedback if there really is a strong use-case for it to stay).

The main reason we're looking at removing it is that while it makes writing samples easy, it's risky to use for competition because it's far too easy to write code that doesn't correctly transition between modes (e.g. auto to teleop) or not update motor values frequently enough.  To use an analogy, while we can't prevent foot-shooting by teams, we can limit the damage.

The only reason more experienced teams used SampleRobot (that we're aware of) was to have a timed, rather than packet-based, iterative loop--with the addition of TimedRobot this should no longer be a concern.

rlee287, 2018-01-09:
My team has developed our own Java framework at https://github.com/frc604/robotnik2-framework-public. The central class at https://github.com/frc604/robotnik2-framework-public/blob/master/src/com/_604robotics/robotnik/Robot.java extends SampleRobot and creates its own iterative loop.

Kevin-OConnor, 2018-01-09:
This seems to be pretty thoroughly discussed in #472. The deprecation scares off teams that shouldn't be using it (in a way that no documentation can possibly do) while allowing teams to continue using it if they understand the risks.

closed 2018-01-09 by Kevin-OConnor