wpilibsuite/allwpilib #881: PIDCommand usePIDOutput(double output) - output remains NaN

VitruvianBots, 2018-01-09:
Trying to re-implement PIDCommand using the 2018 wpilibj appears to not work. Testing shows that the PIDController is properly enabled when running, but output remains null (NaN).

Code:
```
public class DriveTurnWithGyro extends PIDCommand{
    static double kP = 0.1;        		// Start with P = 10% of your max output, double until you get a quarter-decay oscillation
    static double kI = 0.001;           // Start with I = P / 100
    static double kD = 0;           	// Start with D = P * 10
    static double period = 0.01;
    
    double throttle, setpoint;
    Timer stopwatch;
    boolean lock = false;
    
    public DriveTurnWithGyro(double speed, double angle){
        super("DriveTurnWithGyro", kP, kI, kD, period);
        getPIDController().setContinuous();
        getPIDController().setAbsoluteTolerance(0.1);
        getPIDController().setOutputRange(-1, 1); 
        
        this.throttle = speed;
        this.setpoint = angle;
    }
    
    @Override
    protected double returnPIDInput() {
        // TODO Auto-generated method stub
    	return Robot.driveTrain.spartanGyro.getAngle();
    }
    
    @Override
    protected void usePIDOutput(double output) {
        // TODO Auto-generated method stub
        DriverStation.reportError("Using Output: " + output, false);
    }
    
    // Called just before this Command runs the first time
    protected void initialize() {
        // Use this to test/tune PID loop through smartDashboard
    	//SmartDashboard.putNumber("kP", kP);
        //SmartDashboard.putNumber("kI", kI);
        //SmartDashboard.putNumber("kD", kD);
    	
        kP = SmartDashboard.getNumber("kP", kP);
        kI = SmartDashboard.getNumber("kI", kI);
        kD = SmartDashboard.getNumber("kD", kD);
                
        Robot.driveTrain.spartanGyro.reset();
        stopwatch = new Timer();

        getPIDController().setSetpoint(setpoint);
        getPIDController().setEnabled(true);
        DriverStation.reportError("Initialized", false);
        
    }
   
    // Called repeatedly when this Command is scheduled to run
    protected void execute() {
    	SmartDashboard.putNumber("PID Output", getPIDController().get());
    	SmartDashboard.putNumber("Delta Setpoint", getPIDController().getDeltaSetpoint());
    	SmartDashboard.putNumber("Setpoint", getPIDController().getSetpoint());
    	SmartDashboard.putBoolean("On Target", getPIDController().onTarget());
    	SmartDashboard.putBoolean("Enabled", getPIDController().isEnabled());

    	SmartDashboard.putNumber("Stopwatch", stopwatch.get());
    	SmartDashboard.putBoolean("Lock Value: ", lock);]
    }
    
    
    @Override
    protected boolean isFinished() {
    	if(getPIDController().onTarget() && !lock) { // When you are in range && you are not locked
    		stopwatch.start();
    		lock = true;
    	} else if(!getPIDController().onTarget() && lock){ // When you are outside of range && you are locked
    		stopwatch.stop();
    		stopwatch.reset();
    		lock = false;
    	}
    	
    	return stopwatch.get() > 1;  	
    }
    
    // Called once after isFinished returns true
    protected void end() {
        Robot.driveTrain.setDriveOutput(0, 0);
    }
    
    // Called when another command which requires one or more of the same
    // subsystems is scheduled to run
    protected void interrupted() {
        this.end();
    }
```

VitruvianBots, 2018-01-10:
Okay, so further testing shows that if the .setContinuous() function is called (or set to true) causes issues in both PIDCommand and PIDController. Setting it to false or removing the function gets the PIDController to run.

sciencewhiz, 2018-01-11:
With setContinuous, you need to set an input range.

PR #883 (2018-01-12): Fix PIDController with Continuous and no Input Range set.
Fixes #881
closed 2018-01-12 by PeterJohnson