wpilibsuite/allwpilib #882: Problem installing Linux frc-toolchain without the beta ppa

kjaget, 2018-01-09:
Hopefully this is the correct repo to track this - if not let me know and I can move it.

We're seeing a problem installing the Linux C++ toolchains on new machines.  The problem comes when runn ing :
```
sudo apt-add-repository ppa:wpilib/toolchain
sudo add-apt-repository ppa:webupd8team/java
sudo apt-get update
sudo apt-get install git libc6-i386 curl jstest-gtk gradle oracle-java8-installer frc-toolchain meshlab cmake libprotobuf-dev libprotoc-dev protobuf-compiler
```

The error is 
```
The following packages have unmet dependencies:
frc-toolchain : Depends: g++-arm-frc-linux-gnueabi (>= 5.5) but it is not going to be installed
               Depends: libc6-frc-armel-cross but it is not installable
               Depends: libc6-dev-frc-armel-cross but it is not installable
               Depends: linux-libc-dev-frc-armel-cross but it is not installable
               Recommends: frcmake but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
```

Adding the toolchain-beta ppa, followed by an apt-get update fixes the problem.

Thanks for checking this out!

ThadHouse, 2018-01-09:
It's a known launchpad issue. Their build system is backed up, causing no new builds to actually be built. Once their system comes back the beta ppa will no longer be needed.

kjaget, 2018-01-09:
OK, thanks. We'll proceed with the beta toolchain for now - doubt there are significant differences at this point.

AustinShalit, 2018-01-09:
Here is the announcement regarding the downtime: https://lists.ubuntu.com/archives/launchpad-announce/2018-January.txt

AustinShalit, 2018-05-30:
Closed by https://twitter.com/launchpadstatus/status/956033098860453894

closed 2018-05-30 by AustinShalit