wpilibsuite/allwpilib #885: Cannot report errors on wpilib.screenstepslive.com

eforbes, 2018-01-12:
Not sure where to report this but the "Report Errors" form on wpilib.screenstepslive.com seems to be broken. Filling out the form and clicking submit does not do anything. Clicking submit again causes a "reCAPTCHA verification failed, please try again." error message to show up and resets the reCAPTHA, even though it was successfully checked.

Kevin-OConnor, 2018-01-12:
Filling out the form and clicking submit submits the form (Your comment was submitted). The comment does not appear because the comments are moderated (and generally not posted, we usually make the fix and delete). This is a built in behavior of the Screensteps site and not something we can change.

closed 2018-01-12 by Kevin-OConnor

eforbes, 2018-01-12:
Thanks, glad to know it's working and it's unfortunate that there can't be any feedback to show that it has been submitted.