wpilibsuite/allwpilib #889: FRC Message Console (Driver Station option) causes loss of communications to roboRIO

tom131313, 2018-01-14:
FRC Driver Station - version 18.0 / View Console / FRC Message Console apparently is the cause of the Driver Station to lose connection with the roboRIO and indicate no robot code running within a minute or two of opening the console.

A substantial amount of Java console output can be displayed indefinitely on the Eclipse Riolog (several hours of flawless operation).

However, if using the Driver Station FRC Message Console, the Driver Station claims lose of connection to the roboRIO and no robot code within a minute or two of opening the Message Console Window. The terminal to the roboRIO (say using PuTTY) still connects just fine and the user robot program appears to still be running (as viewed by ps or top commands).

The only way to re-establish Driver Station connection to the robRIO is to reboot the roboRIO.

calcmogul, 2018-01-14:
Did you use Riolog at the same time as the Driver Station FRC Message Console? We've experienced this issue intermittently and don't yet know whether it's Riolog or something else causing it. We'd appreciate it if you could attempt reproducing the problem with Riolog in the opposite state of openness. For science.

tom131313, 2018-01-14:
Yes you are on to something about the Riolog and the FRC Message Console
not playing nice together.

I've run a few hours more of tests and find some that do not fail (in the
hours of testing) and one that fails every time within two minutes.
==============
Let's assume a fresh start:
Boot the PC (probably don't need that fresh of a start, though)
Start Eclipse and Riolog (mine shows automatically in the initial
perspective, now)
Start Driver Station
Enable robot
This configuration will run indefinitely

Later while robot is still running add to this the Open Gear / View
Console / FRC Message Console displays.
This too will run indefinitely with both logs/consoles running at once 
==============
Similarly start the FRC Message console first instead of starting
Eclipse/Riolog first:
Let's assume a fresh start:
Boot the PC (probably don't need that fresh of a start, though)
Start Driver Station
Open Gear / View Console / FRC Message Console displays
Enable robot
This configuration will run indefinitely

Later while robot is still running add to this Start Eclipse and Riolog
(mine shows automatically in the initial perspective, now)
This too will run indefinitely with both logs/consoles running at once
==============

Now the case that fails:
Through any gyrations of development/testing (starting and stopping
Driver Station / robot / Eclipse) that results in this state:
Eclipse Riolog has robot output already showing in it - that is - the
Riolog's initial help message ("This is the robot console....") has been
overwritten by robot output
and the FRC Message Console is not open.

Then open the FRC Message console and enable the robot to produce robot
messages on both the Riolog and FRC Message Console.

The Driver Station will claim loss of communications / no robot code in
two minutes.

Rebooting the roboRIO and Enable will again produce robot messages on
both the Riolog and Message Console for two minutes give or take a couple
of seconds then again the Driver Station claims no communications / no
robot code.

This error state can be reproduced forever (every few minutes - roboRIO
reboot, enable robot, fail to communicate within two minutes) until
either the Riolog or the FRC Message Console is closed.  If both robot
messages windows are desired then the Riolog may be reopened after it is
closed because it's okay to run the FRC Message Console if the Riolog has
been started from its initial state - "This is the robot console..." help
message showing when the FRC Message Console is opened.

Warning:
The usual user/computer analyst relationship warnings apply - "Believe
everything the user says;  Believe nothing the user says."  These tests
over the past few days were tedious and confusing to me as a dozen
hardware components were swapped out and  hundreds of buttons
pushed/clicked.  You still might not have the whole story but I'm sure we
are getting close.  I see others on a Chief Delphi thread (to which I
contributed) have similar problems.  Note that I am NOT using a radio -
only wired

Thanks,

Rick
=====================

Richard K. Thomas

mikagika, 2018-01-15:
We definitely haven't done as much debugging, but we (team 4611) have experienced the problem from our driver stations that don't have Eclipse installed. 

Again, with somewhat limited testing, it seemed like using the 2017 driver station may not exhibit the problem, even when used with the same update for 2018 RoboRios and radios. However, that may be because we rebooted, then connected with that machine and didn't start eclipse/Riolog anywhere first.

Thinking along those lines, it's possible when we've had the problem with a dedicated drive station, it's possible that after the reboot a coder first connected to push code before we connected the drive station. Although what trouble that causes that breaks the connection for the driver station laptop that doesn't even have Eclipse installed, I couldn't guess.

We have a meeting tomorrow, we'll try to do some more testing paying closer attention to things. As it stands, it's really difficult to get any meaningful driving testing done with these disconnections.

tom131313, 2018-01-15:
The steps I described above for the failure are vague.  Here's a better procedure to failure with the pertinent steps.

Boot the PC (probably don't need that fresh of a start, though)
Start Eclipse and Riolog (mine shows automatically in the initial
perspective, now)
Start Driver Station
Open Gear / View Console / FRC Message Console displays
Reboot roboRIO (required step toward failure - can use D.S. button or roboRIO reset button)
Enable (required step after the required reboot - no failure waiting in Disabled mode)
Soon it will fail.  Yesterday it was every 2 minutes and today it's every 90 seconds.

Except for the initial required start of Eclipse/Riolog and Driver Station/FRC Message Console, it doesn't matter how often they are started and stopped - restarting them isn't necessary and was done only out of our ignorant desperation.

Rebooting the roboRIO while the logs are open seems to be key to the failure.  Enable after reboot is required to fail.

This error state can be reproduced forever (every few minutes - roboRIO
reboot, enable robot, fail to communicate within a minute or two minutes) until
either the Riolog or the FRC Message Console is closed.  If both robot
messages windows are desired then the Riolog may be reopened after it is
closed because it's okay to run the FRC Message Console if the Riolog has
been started from its initial state - "This is the robot console..." help
message showing when the FRC Message Console is opened.

mikagika, 2018-01-15:
So far today, we have found and renamed the 2018 RioLog jar file and when using machines that have had that done to them, we seem to have no comm issues. The RioLog comes up with a red X in eclipse, but we haven't seen any other fallout from not having the RioLog jar file. (It was in a .p2 folder somehwere.) 

We also added printLns to the code so we were generating a fair bit of console traffic. 

We had a coder connect with a machine that had RioLog active and we lost comms shortly after enabling from the drive station.

Another coder working on a different chassis connected first with drive station, made sure she had comms, but then went opened eclipse and pushed code with RioLog active, then went back and enabled from the drive station and had no comm problems after that. 

I wouldn't guarantee that these testing results have been perfectly controlled, but I do have a reasonable level of confidence around them. I would summarize them as:

A) If you don't care about RioLog, renaming the jar file seems to prevent comm problems. So it seems that the RioLog is somehow related to the problem. 
B) Connecting from drive station first seems to also avoid the problem. However, when developing code, it does require a bit more discipline to remember to open drive station first.

tom131313, 2018-01-15:
I  lied about needing to Enable the robot from the Driver Station - I had a failure while Disabled.

Again, I can have the Riolog or the Message Console running but if I want to run both at the same time then I can't reboot the roboRIO without getting the failure (unless I turn one of logs/consoles off first).

Renaming the riolog.jar certainly prevents it from running but just not having it show in a view suffices (closed and that state persists until shown again).

I'll try testing with a Driver Station PC different than an Eclipse/Riolog PC.

jhershbe, 2018-01-15:
Are you certain that you need to open the console window on the DS? I would think that simply having the DS connected to the robot would be the same.

mikagika, 2018-01-16:
I'm not 100% sure, but our coder who opened the drive station first today and then worked fine without problems, probably didn't open the console window. I didn't ask her to, and being new to the team it's probable that she wouldn't have thought to open the console window.

tom131313, 2018-01-16:
Not only does the Riolog not play nicely with FRC Message Console but there is connection failure with Riolog and the Shuffleboard open at the same time.  And that's with Shuffleboard on a different PC than the Riolog.  I had no problem with the Message Console closed and Shuffleboard closed and the Riolog open but I'll test more tomorrow making sure I open the DS first.

tom131313, 2018-01-16:
Here's what's wrong with the Riolog/Message Console - the high-level view that (almost) explains what I'm seeing after many, many tests.

The failures occur on a timer and are an interaction between two (seemingly) independent tasks even across a network of two PCs (and a roboRIO).

The timings are (almost) the same as the Driver Station Setup Practice Timing (s).  We've had failures at nearly 15 seconds, 90 seconds, and around 125 seconds.  (Yesterday I had three test failures in a row at 90.1, 90.2, and 90.15 seconds of the timer that I set.)

Lacking source code and analyzers I have a naive view of networking.  That is I don't know how the following statement can be true but this is what it looks like to me.  It appears that the Riolog reading of the packet causes the Driver Station Message Console reading of the packet to be delayed enough to cause the Driver Station to miss sending a stay-alive message to the roborRIO if the Driver Station is also processing one of its (Practice) timers.

mikagika, 2018-01-16:
Just to note, we've had the problem even if the laptop that connected to the roboRIO first, then disconnected from the network. It almost seems like the Riolog is causing something running on the roboRIO itself to do something untoward because the driver station will disconnect even if it's the only device connected to the roboRIO's network.

tom131313, 2018-01-16:
Maybe there is network contention causing any of the devices to respond slowly sometimes - I don't think I have the means to measure that.

I am also somewhat concerned about the original Chief Delphi title regarding radio disconnects.  I have never used a radio in my testing.  Maybe there are multiple problems.

I'm up for more testing if someone has anything for me to try.  I don't have the source code - don't know where to find it and probably no way to compile it.  If the maintenance programmers want to add some "instruments" to the code, I'd be glad to test some more.

mikagika, 2018-01-16:
Agreed, while we usually are connected via radio we had seen disconnects when connected directly via Ethernet. I'm not sure about USB. One of my initial throughts was that it was a wireless network contention issue somehow, even though we were trying to keep everybody off from the robots' networks unless they were actually testing. But then the disconnects over ethernet kind of blew that idea out of the water.

And renaming that .jar file yesterday made everything just work like it used to. Other than the red x showing up when Eclipse opens the rioLog, which we don't really care about.

tom131313, 2018-01-16:
We had the same disconnections using the roboRIO USB ports.  Only used USB because it was still wired for the imaging process and too lazy to change to the other port.  With the failure we moved immediately to the Ethernet.  Alas, that did not help.

I decided to learn something this morning (the hard way, of course) and downloaded Microsoft Message Analyzer.  Captured hundreds of thousands of packets including the failure.  Thought I saw something in the last packet before discount but turned out to be rare but not unheard of.

Also found Riolog in GitHub.  Makes one long for the days of NetConsole and UDP port 6666.  There are a few helpful comments in it but lacks overall view of how it works (slow going to reverse engineer).  It makes one connection to the Driver Station TCP 1742 at the beginning for info about the roboRIO and connects to roboRIO TCP 1741 using multiple addresses and I think more than one of those connections could be made (using two different names for the same address).

PeterJohnson, 2018-01-17:
Thanks for the reporting.  We believe we've identified the issue and have a fix in the works that's undergoing internal testing right now, so there's no need to continue trying to diagnose it.  We're trying to get the fix out to everyone as soon as we can, but in the meantime you should throttle your console output (e.g. output only 1 of 10 teleop loops or similar).

If you have questions about the riolog code, I'm the one who wrote it if you have questions (yes, I'm not always the best at commenting).  Note the riolog repo is not 100% in sync with the Eclipse version (which is in the EclipsePlugins repo).  The multiple connection attempts code resolves IPs and won't try to connect to the same IP twice (and in any case the first connection wins and the others are discarded).

mikagika, 2018-01-17:
Great! Glad to hear it! 
Is there any particular reason why the riolog was set to open automatically this year?

tom131313, 2018-01-17:
Fantastic.  Looking forward to the new version.  Thanks.

PeterJohnson, 2018-01-17:
One thing that could help is if you can capture a core dump of FRC_NetComm on the roborio when this occurs, it would help confirm there's no other latent bugs causing the issue.  The way to do this is to use opkg to install gdb (`opkg update; opkg install gdb`), then use the `gcore PID` command (where PID is the FRC_NetComm process).

PeterJohnson, 2018-01-17:
The automatic opening of riolog was actually implemented before we switched over to the new TCP console.  Riolog was significantly underutilized (many teams didn't even know it existed) and we thought it would be helpful to teams to have it automatically show riolog on successful deploy so teams could see if their code crashed, etc.

rrossbach, 2018-01-17:
Here are core dumps - I dumped both processes that referenced the NetCommDaemon.  FYI I also put these up in the CD thread since Joe and I were messaging each other on CD as well.....
[NetComm-coreDumps.zip](https://github.com/wpilibsuite/allwpilib/files/1640961/NetComm-coreDumps.zip)

mikagika, 2018-01-17:
Yeah, we never used riolog AFAIK, but I don't know that we missed it--we always used the console in the driver station. Maybe there's some benefit to having it there in Eclipse that will be useful. But depending on what phase we're in, there's also the situation of immediately disconnecting after deployment so we can connect the driver station. 
But we are intending on using the new Shuffleboard this year to allow the drivers to change some of the things that we were manually changing in the code, necessitating a redeploy so maybe we'll have less of that this year.

jhershbe, 2018-01-19:
The issue in Ross's dump matches the issue I fixed. Moving forward with release.

schleprock, 2018-01-22:
i noticed a new release of eclipse wpilib yesterday (21-jan 2018), was the fix included in that release? if not, is there an ETA for a fix?

dbadb, 2018-01-25:
connecting potentially related thread:

https://github.com/robotpy/pynetconsole/issues/7

Would love to hear experts respond to whether this is likely the same issue.

Thanks!

jhershbe, 2018-01-26:
The update is released.

http://www.ni.com/download/first-robotics-software-2017/7183/en/

schleprock, 2018-01-27:
are you sure it's been released? looking at the download site the version is still 2018.1.0 and the released date is: 
01-03-2018

mikagika, 2018-01-28:
Check again. When I last looked it showed a release date of early this week. In limited testing this evening, I haven't had any random disconnects with the new roboRIO image and the new eclipse plugin.

schleprock, 2018-01-29:
you are correct, the new update is now available and the released date is correct. we successfully installed the new image on our roborio and connected several laptops running the new Riolog in eclipse with many messages showing up in the console and the robotCode did not crash. awesome job!!

timfame11, 2018-02-04:
That didn't work for me because I updates my driver station and I still get crashes. When I close most of the unused programs I work, but sometime I still get random crashes once in a while.

PeterJohnson, 2018-02-05:
Did you update the roboRIO image to v17?

timfame11, 2018-02-05:
Yes

timfame11, 2018-02-07:
I was mistaking because I don't know what version the roboRIO is on.