wpilibsuite/allwpilib #892: Write wrapper for PIDController that implements SpeedController

Oblarg, 2018-01-16:
It's a bit of a chore to implement PID velocity control with WPILib drive objects at the moment.  A simple wrapper that implements SpeedController would greatly help many teams add this functionality without changing much of their code.

calcmogul, 2018-01-30:
The wrapper controls the velocity of one motor via a PIDController and implements SpeedController so it's a drop-in replacement for DifferentialDrive's SpeedControllers. For that to work, the wrapper would have to provide some mapping from -1..1 to the valid velocity ranges of the controller. The user would have to set this beforehand. My main concern with this is making the controller easily swappable. I guess that could be done by passing a controller in, but I'm not sure if it should be a PIDController or just a function that performs the controller calculation. Doing the latter means the function calls can be batched into one Notifier within a DiffDriveController.

By making the wheel controllers a drop-in replacement, they can be omitted if just an outer controller is desired for controlling the robot's pose. I suppose the DifferentialDrive could be composed with a DiffDriveController that calls TankDrive() with the control input. I assume TankDrive() would be easiest to control, but other inverse kinematics are possible.

There are different possible algorithms for DiffDriveController. One example is using one controller for the forward position, another controller for the heading, and superimposing their control inputs for each respective wheel. This method can also control the forward velocity and turning rate.

AustinShalit, 2019-08-04:
Please follow @calcmogul's suggestions.

closed 2019-08-04 by AustinShalit