wpilibsuite/allwpilib #901: PIDController setToleranceBuffer filters loop input

Oblarg, 2018-01-20:
We updated our test bed today to 2018 WPILib, and noticed that our "hold angle" loop was suddenly behaving in a wildly erratic manner.

Upon inspection of the current version of the PIDController object, I notice that setToleranceBuffer now applies a LinearDigitalFilter not just to the error used to calculate onTarget (which would result in similar functionality to the 2017 version), but also filters the input to the PID loop.

We were using a significant tolerance buffer, and so this introduced a large amount of phase lag into our input and caused unstable loop behavior.  While the setToleranceBuffer method is currently deprecated, this change thus invisibly breaks a potentially very large quantity of legacy code.

calcmogul, 2018-01-21:
You're right that there's a phase lag in the controller that wasn't there before. The problem with the old approach is that the error used by the derivative term isn't filtered by the tolerance buffer, which is one of the main reasons people would (erroneously) attempt to use the tolerance buffer. I'd argue the previous behavior was a bug in PIDController, but it didn't affect anything because the D term isn't used in the velocity PID implementation.

The controller should be operating on the same input as the OnTarget() calculation (that is, the controller should be using the same information provided by the estimator). If the input isn't clean enough for error calculations as-is, that garbage data shouldn't be used by the controller either. If a better state estimate with less phase loss is desired, consider using an estimator based on an actual system model. See http://frc971.org/workshops for 971 showing how to do this with an elevator. I admit this is really a non-solution to this issue though.

I'm curious, why would you be using a buffer that large? If the input is oscillating that much all the time, you probably need a larger tolerance band around your reference.

Oblarg, 2018-01-21:
We weren't actually "using" a buffer that large for the command in question (a "hold angle" loop), because the command in question wasn't ever actually checking onTarget() but had a "vestigial" buffer setting in our config file as a result of some boilerplate copy-pasting from earlier turn-to-angle loops (which did previously use a fairly lengthy buffer to deal with occasional overshoot: back before we implemented custom feedforward on the inner velocity loops to better deal with stiction, it was rather annoyingly difficult to turn acceptably-quickly without overshoot, and the loop tended to exit on overshoot without a rather lengthy buffer, especially because the error values would tend to cancel as the robot swung past the setpoint).

calcmogul, 2018-01-21:
That sounds similar to what the person in #500 did. As mentioned there, the tolerance buffer should not be used for dealing with overshoot. Since PID controllers have a second order response, OnTarget() should at least check that the error as well as change in error are within tolerance. The correct solution is checking all system states are within tolerance, but that requires full-state feedback.

A defaulted argument could be added to SetTolerance() that specifies a tolerance for the change in error. I did that in #544 but need to port it over to the current PIDController implementation.

Oblarg, 2018-01-21:
That sounds like a good change.

I'll suggest that it might be better to simply remove setToleranceBuffer rather than to deprecated it along with an invisible and potentially-breaking change, and to rename the current function to something more-descriptive.  At this point, the method does something significantly different than what it did before, and this would make this fact much clearer to users.

Better to explicitly break code that might be otherwise invisibly broken, in my opinion.

calcmogul, 2018-01-21:
The plan is to remove it completely in favor of external filters for composability and customizability. However, our API change/removal policy requires that the original feature be deprecated for one year so teams have time to migrate. I agree that breaking code loudly is preferred, but the original intent was to maintain the same behavior until removal. We just messed up the shim implementation for doing that.

I'll see what opinions some other devs have on fixing this issue, because I can't think of a good way of fixing this without affecting our future direction. I guess adding more to the deprecation message to mention this issue is an option?

brettbbonner, 2018-01-30:
Is there any further information available as to whether and when the fix committed above will be released in WPILib?  We're trying to figure out if that fix will be available to teams, or if we need to code around the change ourselves.  Thanks.

calcmogul, 2018-01-30:
Adding the second argument to SetTolerance() is done in #908, which is set for 2019. #902 is set for update 3, so that should be merged in a week or two. Feel free to use the files from the PRs until then.

AustinShalit, 2019-08-04:
Closed by #1300

closed 2019-08-04 by AustinShalit