wpilibsuite/allwpilib #905: Differential Drive Error

GaganBhat, 2018-01-21:
Hey,

Everytime I used differential drive, I get the not outputted enough error in the driverstation. I have tried placing this in the robotInit, and in a subsystem constructor. Multiple people have the same issue when using differentialDrive. 

Regards
Gagan Bhat

Kevin-OConnor, 2018-01-21:
If you are only updating the output in a constructor or robotInit, that is not often enough. RobotSafety (which is enabled by default for the drive classes) requires you update the output every 100ms in order to remain enabled. 

For a command based robot this means you should be updating the value in the execute method of commands that use the subsystem, and you should have a default command for the subsystem.

The other option is to disable RobotSafety for that object, but I do not recommend that.

GaganBhat, 2018-01-23:
I updated the output every 100ms and tried to disable safety, but both resulted in this error.
[image]

Kevin-OConnor, 2018-01-23:
That means that the SpeedController object you are passing in is NULL. You need to make sure your SpeedController is created before you pass it into the DifferentialDrive object constructor.

GaganBhat, 2018-01-23:
You are right, sorry I overlooked that. Thanks a ton!

closed 2018-01-24 by 333fred