wpilibsuite/allwpilib #907: FRC Radio Configuration Utility on non-English Windows

moesoha, 2018-01-23:
The tool can't run on most of non-English edition of Windows. Interface list is null. I guess it is because of the default code page.

[image]
*Here's the problem!*

When I changed the setting "Language for non-Unicode Programs" in Windows locale setting to `en_US`, this works fine. But this way is inconvenience.

[image]
*Works fine now!*

I want to solve the problem but I found the tool is closed source. The problem affects a lot of teams I think, so can you give a official fix?

Kevin-OConnor, 2018-01-23:
This is a known, documented issue that will not be fixed for this season. I did just update the documentation to reflect your suggestion of setting the Locale setting as a workaround. Thanks!

rdelfin, 2018-02-06:
Is there any way we we could write a PR for a proposed fix? I can't seem to find the associated code in this repo. It's also been an issue with a lot of FRC teams here in Mexico for some years and if there's anyway we can help get it fixed, we'd be glad to.

moesoha, 2018-02-06:
@rdelfin I think the radio configuration utility is a closed source project, but don't worry about it. This issue is tagged with milestone 2019, i think this'll be solved next year.

And the solution i mentioned above was added to the [official document](http://wpilib.screenstepslive.com/s/currentCS/m/getting_started/l/144986-programming-your-radio).

rdelfin, 2018-02-06:
I get what you're saying. We're seeing the question on this issue prop up a lot this year and unfortunately for a lot of the teams here. Since most of the Windows installations here in Mexico come in single language editions, the fix doesn't seem to work (as tested by @franspaco). For the time being, he's looking into alternatives.

moesoha, 2018-02-06:
@rdelfin i remember that **"Language for non-Unicode programs"** can be changed in every Windows edition. And here in China, no team which met the problem cannot solve it in this way.

rdelfin, 2018-02-06:
We did give it a try and still got the same issue. For some reason, even after changing the language settings and restarting as instructed, the terminal stayed in Spanish. Our best guess right now is that either the fact that they're single language editions make the setting moot, or that it has something to do with the fact that the entirety of the Spanish keyboard (including the characters áéíóúñ) exists under (extended) ASCII, but we're not sure, since this seems so weird.

moesoha, 2018-02-06:
@rdelfin The interface name in the control panel should contain only ASCII chars

rdelfin, 2018-02-06:
@moesoha the issue isn't because it can't pick up non-ASCII characters. Rather, when the Radio configuration utility runs `ipconfig /all` on a terminal, the output of said terminal is in Spanish. We've been unable to change this language. From what we can tell, it parses the output by looking for things like:
```Ethernet adapter vEthernet (Default Switch)```
However, it is instead getting:
```Adaptador Ethernet vEthernet (Default Switch)```
It just seems some editions of Windows cannot change the language of the terminal, which this tool needs.

moesoha, 2018-02-07:
@rdelfin In Chinese edition of Windows 7 Home Basic, `ipconfig` returns English result. Maybe this is different between different languages.

[image]

rdelfin, 2018-02-08:
@moesoha Indeed, I think Microsoft translated the terminal for some languages, and for others it didn't. I think the only thing left to do is wait for a fix in that case.

moesoha, 2018-02-09:
@rdelfin I wrote a program to mock the system `ipconfig`. Interfaces can be list with this program, but I don't have a radio to flash. Could u please try this hack?

[image]

Just put the `ipconfig.exe` in the zip below into `C:\Program Files\FRC Radio Configuration Utility` or `C:\Program Files (x86)\FRC Radio Configuration Utility` and run the utility normally.

[ipconfig.zip](https://github.com/wpilibsuite/allwpilib/files/1710529/ipconfig.zip)

moesoha, 2018-02-10:
@Kevin-OConnor The program *(more info: [moesoha/frc-rcu-language-fixer](https://github.com/moesoha/frc-rcu-language-fixer))* i mentioned above was tested on French and Chinese Windows. It can fix the problem by adding just one exe to the program installation directory.

franspaco, 2018-02-12:
@moesoha I must say, that is a beautiful hack. I was thinking of meddling with the jars in \libs, but that would make it harder to make a general solution for other languages. Highjacking the call for ipconfig is probably the best solution given how the program currently works.

(This also makes me wonder why java's NetworkInterface.getNetworkInterfaces() is not enough, why is there a need for an ipconfig at all? But well, I guess the devs will take care of that for next year)

Kevin-OConnor, 2018-11-06:
@franspaco Ipconfig is needed because Java cannot set the IP of an interface without making a system call. This requires the Windows name of the address which is unfortunately not the name returned by either getName() or getDisplayName()).

@moesoha I have evaluated some alternative approaches, but it looks like using your executable may be the simplest. Based on the license I technically don't need to ask, but are you ok with me distributing your middleware for 2019?

franspaco, 2018-11-06:
Thanks for the clarification @Kevin-OConnor :)
I played with the tool a bit and did manage to fix the bug but only for Spanish so, I agree with your idea.

@moesoha you would greatly help every international team if you agree!

moesoha, 2018-11-06:
@Kevin-OConnor obviously, the answer is YES.

I'm so happy that exec file can help FRC teams.

Kevin-OConnor, 2019-01-08:
Version of ipconfig stand-in from @moesoha distributed with the 2019 radio kiosk should resolve this issue.

closed 2019-01-08 by Kevin-OConnor