wpilibsuite/allwpilib #910: error: type 'Encoder' could not be resolved

woolfelucas, 2018-01-23:
Our team created a cpp project before we installed the updates, toolchains, and plugins for the 2018 season. After we installed all of it, this project updated itself just fine. Here is a sample of our robot code from that project as it stands today.

[image]

However, there was an issue with an std function, which prevents us from outputting updates from our sensors, such as encoders. After some research we found that our issue was caused by how the project was initially built, and that we should try creating a new one. And so we did.

Creating the new project solved the std issue. However, a new issue came up wherein the Encoder type is not recognized at all. Below is a picture of the same sample of the same robot code, in a different project, though with identical libraries, includes, and library paths.

[image]

As you can see, the Encoder type is not being recognized, even though other WPILib types (and 3rd party types) are being recognized. The error is: "type 'Encoder' could not be resolved".

I've created 4 different projects, one of which was after I simply repaired the toolchain, another of which was after a reinstallation of everything (eclipse, FRCUpdateSuite, FRCtoolchains, PhoenixFramework, and FRC Plugins, in that order). I also tried migrating the plugins to a new workspace, wherein one of the projects was initialized. Yet, the same error is occurring for each project and the robot code cannot recognize the Encoder type, and only that.

I can't find any documentation on this issue. Any Ideas?

-Lucas, controls lead, FRC4039

PeterJohnson, 2018-01-23:
This is probably not actually an error, but rather Eclipse is reporting a false positive in its Intellisense (interactive real-time assessment of errors).  If your project can compile successfully it’s not actually an error.

woolfelucas, 2018-01-24:
Thanks! I can't believe I forgot about that.

closed 2018-01-24 by woolfelucas