wpilibsuite/allwpilib #917: Return this in setInverted

twinters007, 2018-01-26:
This would allow one to do 

`foo = SpeedControllerGroup(Victor(0).setInverted(True), Victor(1))`
which could be useful for teams

calcmogul, 2018-01-26:
This is inconsistent with our current API (namely that we don't use builder objects anywhere). Could you explain how this could be integrated into WPILib's API in a more holistic manner?

twinters007, 2018-01-26:
Yeah I guess it'd be weird to only do it in setInverted. Maybe for 2019 we could declare certain classes as 'builder objects'.

auscompgeek, 2018-01-26:
I'm looking at this suggestion, and all I can think of is "this isn't JavaScript".

twinters007, 2018-01-26:
Maybe javascript is on to something 🤔

auscompgeek, 2018-01-27:
Method chaining is a pattern in JS because of jQuery, where it mostly made sense to return the DOM element since one would need to set a _lot_ of things using various different methods.

This isn't the case here, and method chaining isn't really a thing in C++/Java outside the builder pattern.

PeterJohnson, 2018-09-28:
We're not planning on doing this without a strong argument to do so, so closing.

closed 2018-09-28 by PeterJohnson