wpilibsuite/allwpilib #919: I2C Missing 2 Bus

ProfTechnical, 2018-01-27:
Our team is working with the LIDAR Lite 3, and we have not been able to get any communication with the sensor. We SSHed into the RoboRIO and started looking at what the sensor was sending. It was active and had information being sent over different registers. The issue that we now have is that the information is being sent over bus 2. The LIDAR is plugged into the Onboard I2C port, but there is no way for us to directly get information over code, since the only ports used in the I2C constructor are:
```
public enum Port {
    kOnboard(0), kMXP(1);

    @SuppressWarnings("MemberName")
    public final int value;

    Port(int value) {
      this.value = value;
    }
  }
```
Nowhere is the 2 Port available for use. Please add a kInternal(2) Port to the Port enum so teams can access I2C devices, even if they are on an unexpected bus.

PeterJohnson, 2018-01-27:
If you're referring to the "port 2" Linux device (e.g. /dev/i2c-2), kOnboard is the correct one to use.  kOnboard maps to /dev/i2c-2, and kMXP maps to /dev/i2c-1.

ProfTechnical, 2018-01-31:
So why is it that the (0) value is being sent into the I2CJNI rather than 2? Inconsistency from Java to Linux?

AustinShalit, 2018-01-31:
Take a look at this piece of code: https://github.com/wpilibsuite/allwpilib/blob/master/hal/src/main/native/athena/I2C.cpp#L57-L66

Because the mapping is at the HAL layer, this is true for C++ too.

ThadHouse, 2018-05-14:
Did you ever get the device working. We don't know of any I2C specific issues that would cause your issue.

PeterJohnson, 2018-07-27:
Closing.  Please reopen if you are still having an issue.

closed 2018-07-27 by PeterJohnson