wpilibsuite/allwpilib #920: "Unable to access jarfile FRCUserProgram.jar"

dawilson9, 2018-01-27:
We're trying to bring up our 2018 FRC code but, regardless of the Java application we build and deploy, we are always seeing "Unable to access jarfile FRCUserProgram.jar" in the Eclipse RIOLog window after the code is successfully copied to the robot. We initially tried with our 2017 code ported to the 2018 WPILib/CTRE_Phoenix but then moved back to a completely empty skeleton generated with Robotbuilder in case something we were doing was messing things up. 

So far, we've SSH'd into the RoboRio and verified that /home/lvuser/FRCUserProgram.jar" exists. We've also copied the file back via SFTP and verified that it is identical to the file generated by Eclipse. We also tried cmod'ing the file in the RoboRio file system to 777 in case it was a permissions issue. We've also reimaged and rebooted the RoboRio a couple of times. No luck.

At this point, we're stuck. Can anyone offer any suggestions on the next step in diagnosing this problem?

dbadb, 2018-01-27:
Our team encountered something similar, not sure if this applies in your case.   There is a file copied during deployment called robotCommand (or robotCommandDebug).  Through, slightly odd circumstances, that file had a CRLF introduced by git.  Since this file is parsed in linux-land, that's a bad character and the filename produced during the robot bootstrap process is bogus. This was where the FRCUserProgram.jar message came from.

dawilson9, 2018-01-28:
Excellent info - many thanks! I managed to get into the robot from my Mac this afternoon via WiFi (Mac access is causing me a whole range of other problems) and verified that the robotCommand file was indeed corrupted with a non-printing  character at the end of the line. I've fixed it and will update as soon as I get the driver station up and running.

closed 2018-01-29 by dawilson9