wpilibsuite/allwpilib #922: TestScheduler: Allow more easily unit testing commands

dcowden, 2018-01-28:
I'm a First Mentor for Team 281 in Greenville, SC. I'm also a full-time enterprise java developer for my day job.

Command-based programming does a really good job of isolating the robot logic into small parts, and really helps keep the code simple.  But I've been really disappointed with the wpilib's structure as it relates to unit testing.  Even though robots are small and relatively simple, my objective in FIRST isn't to code a robot-- it is to teach a solid  understanding of how to be a good programmer.  

It would be really nice if it were possible to run unit tests for commands more easily.  Commands generally reference subsystems, and it is pretty easy to mock those out, so that we can test that Commands call the expected subsystem methods.

But when we want to test combinations of commands, I'd like to be able to run the commands using the scheduler, so that I can test that they will produce the desired result.  Today, this is not possible, because when we instantiate a Scheduler instance, we get this trace:

```
java.lang.UnsatisfiedLinkError: no ntcore in java.library.path
	at java.lang.ClassLoader.loadLibrary(ClassLoader.java:1867)
	at java.lang.Runtime.loadLibrary0(Runtime.java:870)
	at java.lang.System.loadLibrary(System.java:1122)
	at edu.wpi.first.networktables.NetworkTablesJNI.<clinit>(NetworkTablesJNI.java:53)
	at edu.wpi.first.networktables.NetworkTableInstance.getDefault(NetworkTableInstance.java:88)
	at edu.wpi.first.wpilibj.livewindow.LiveWindow.<clinit>(LiveWindow.java:41)
	at edu.wpi.first.wpilibj.SendableBase.<init>(SendableBase.java:34)
	at edu.wpi.first.wpilibj.SendableBase.<init>(SendableBase.java:24)
	at edu.wpi.first.wpilibj.command.Scheduler.<init>(Scheduler.java:92)
	at edu.wpi.first.wpilibj.command.Scheduler.getInstance(Scheduler.java:46)
	at frc.team281.robot.TestScheduler.testScheduler(TestScheduler.java:11)
```
It would be really great if it were possible to instantiate the scheduler without loading the native libraries, because these are typically not available on a build pipeline server like bitbucket or travis ci, and are a pain to install on the developer laptops as well.

What would be IDEAL is a scheduler implementation that allows stepping forward and backwards in time, so that tests can run much more quickly than in real-time. For example, I'd like to be able to set time forward to 5s elapsed, to assert that a command does what I expect.

I've started the process of implementing my own solution, but it is not easy, based on the classes today. I can probably figure something out, but I really think the framework should make it much easier to unit test.

dcowden, 2018-01-28:
Ok, I got further into this study. I started by attempting to create my own copy of Scheduler, preserving as much code as possible.  I took these steps, which are not ideal but are acceptable for now, because they are not needed for unit testing

*  delete the button handling
* delete initSendable()
* delete networktables references
* make a copy of Set and LinkedListElement ( not visible from outside )

but I got stuck at this code:

`	    Enumeration requirements = command.getRequirements();
`

getRequirements() is not visible to my code, so at this point I'm forced to re-write a lot of the core logic :(

dcowden, 2018-01-28:
Another example:
To run commands, RobotState must be set. There's a UnitTestUtility class in the test packages that uses MockRobotState-- I was thinking maybe I could use those, But RobotState.Interface is not accessible from other packages, so its impossible to iinitialize the robot state ( even to a Mock version ) from classes outside of the wpilib code :(

vairamsvsjdo, 2019-01-09:
@dcowden  I don't see a fork repository in your account where you are developing this. I am also interested in such effort. 
Also the following folder has some unit and integration tests
https://github.com/wpilibsuite/allwpilib/tree/master/wpilibj/src/test/java/edu/wpi/first/wpilibj
https://github.com/wpilibsuite/allwpilib/tree/master/wpilibjIntegrationTests/src/main/java/edu/wpi/first/wpilibj

Is it possible to reuse these ideas for this purpose?
@PeterJohnson  I take it this became a low priority with core functionality completed.

What will be a pragmatic way to proceed in creating a testable code with some Mock code?

AustinShalit, 2020-09-13:
I think that this can be solved by adding a robot example of how to write unit tests for robot code.

calcmogul, 2021-11-09:
OBE by simulation support.

closed 2021-11-09 by calcmogul