wpilibsuite/allwpilib #925: SPI configuration methods are confusing and sometimes invalid

dannytix, 2018-01-30:
The names of the methods **setSampleDataOnRising()** and **setSampleDataOnFalling()** in edu.wpi.first.wpilibj.**SPI** (and its C++ counterpart) do not accurately reflect their functions and can lead to an invalid configuration in some cases.

These names are _only_ accurate when the SPI port has been configured with **setClockActiveHigh()**, and have the **opposite** effect when using **setClockActiveLow()**. Under the hood, these methods are just setting the sampleOnTrailing (CPHA) syscall parameter without regard for clock polarity (CPOL) and should be named accordingly.

This _could_ be corrected in the existing methods by adding logic to invert the sampleOnTrailing value when the clock is active low, at the risk of breaking existing user code. Instead, I suggest that **setSampleDataOnRising()** and **setSampleDataOnFalling()** be deprecated, and new methods along the lines of **setSampleDataOnLeadingEdge()** and **setSampleDataOnTrailingEdge()** be added instead. The latter approach would allow existing code to continue functioning, and would bring the class in-line with the way that many SPI devices are documented, making it easier to determine which options are required for a given device. Furthermore, I would suggest adding language to the descriptions of these methods (as well as **setClockActiveHigh()** and **setClockActiveLow()**) that explicitly references how the methods impact the CPHA and CPOL parameters, and/or which SPI modes (0-3) each method supports.

If there is a consensus on implementing one of these approaches, I would be happy to submit a pull request for the change.

ThadHouse, 2018-01-30:
We probably wouldn't merge anything for the 2018 season, as you're right they would be potential breaking changes. However, this definitely looks like something good for 2019. I like the idea of changing to the LeadingEdge and TrailingEdge wording. If we do this between seasons, we could probably end up just removing the old methods, as not many teams directly use those classes, and they'd be simple changes to make. If you want to PR something you are welcome to, or we will end up getting to this over the summer. Thanks for bringing this up though!

dannytix, 2018-01-30:
Makes sense to me. I'll put it on my calendar to check-in after the season ends and build a PR if it's still outstanding.

PR #1056 (2018-05-15): Makes SPI edge changes more obvious
Rising and Falling mean the opposite when active is set high vs low. Leading and trailing makes much more sense.

Closes #925
commit ab70220e (2018-05-15): Makes SPI edge changes more obvious (#1056)

Rising and Falling mean the opposite when active is set high vs low. Leading and trailing makes much more sense.

Closes #925
closed 2018-05-15 by PeterJohnson