wpilibsuite/allwpilib #930: Unhandled exceptions within TimedRobot do not crash program

gerth2, 2018-02-04:
Discovered while testing on robot today. We had an unhanded exception in our autonomous code. Though the error itself was reported in the DS, only the thread running the TimedRobot crashed, and the JVM did not exit. We were expecting the JVM to restart like in previous years ("Robots don't quit!"). 

Based on some quick testing, this behavior is different than IterativeRobot where an unhandled exception crashes back up to RobotBase, and eventually triggers the [System.exit(1) call](https://github.com/wpilibsuite/allwpilib/blob/master/wpilibj/src/main/java/edu/wpi/first/wpilibj/RobotBase.java#L270).

Possible fix: Try/Catch around [the main timed loop](https://github.com/wpilibsuite/allwpilib/blob/master/wpilibj/src/main/java/edu/wpi/first/wpilibj/TimedRobot.java#L30) with appropriate handling and exit on error.

calcmogul, 2018-02-04:
This was fixed in #926.

gerth2, 2018-02-04:
oooh awesome. Dont' know why my search wasn't catching that issue....

gerth2, 2018-02-04:
Checked the PR, that looks exactly like what we were thinking it would be. Looking forward to the release!

closed 2018-02-04 by gerth2