wpilibsuite/allwpilib #937: WPILib Java Deploy not available

pauldennis2, 2018-02-10:
Even after re-installing Eclipse and WPILib, one of our main programmers does not see the "WPILib Java Deploy" option come up (so we have no way to download code to the robot from that computer.

bradamiller, 2018-02-10:
Can you try creating a new Java project with just the defaults and see if you can download that one? Just wondering if there is something with the copy of the project on that one system or something wrong with the release on your computer.

pauldennis2, 2018-02-10:
@bradamiller creating a new Java project works. But when we attempt to download the project from GitHub, the deploy option is missing.

bradamiller, 2018-02-10:
There's a tag in the project file that tells eclipse what type of project it is. I'm guessing that it somehow went away or is incorrect. I asked someone else to look at it who has more experience with the actual plugins than I have, but if you want to get going fast, you could probably drag the files into the new empty working project. Although I'm not sure why doing the pull from GitHub fails on one computer and not on all of them. (I think that's what you are reporting).

pauldennis2, 2018-02-10:
Thanks for passing this along. I've looked at the .project files as well as the build.properties and build.xml files. I can't find any differences between the project I created on my computer (which is able to deploy) and the version of the project from GitHub. We're having the same issue on 2/3 of the computers we've used so far on this.

pauldennis2, 2018-02-10:
Sorry, I should have added that we seem to be having issues with configuring the WPILib JARs (though I'm not sure why this difficulty persists through re-installing. Not sure if this is related. Here are the JARs we have in that project (mentioned above). 
[image]

[image]

[image]

Kevin-OConnor, 2018-02-11:
If the FRC Project Nature shows up in the .project file, even on the machine where the menu option doesn't show, there is probably an exception getting thrown by the plugin. Window > Show View > Error Log will show the error log.

pauldennis2, 2018-02-12:
@Kevin-OConnor When should this exception be getting thrown? in my experience exceptions are triggered by actions, but this is more of a state. Also, what type of exception should we be looking for? We can see a few warnings in the error log, but none is obviously connected to the problem at hand.

pauldennis2, 2018-02-12:
I accept that we are clearly doing *something* wrong in the way we've set up WPILib or Eclipse (or something). I'm confused because we have a bug that we've reproduced across multiple computers & projects that introduces a fatal flaw. Is no one else having this problem?

Kevin-OConnor, 2018-02-12:
I have not seen anyone else reporting a similar problem this season that was not caused by the .project file losing the FRC Nature (typically by not checking the .project into Git). 

The triggers for the menu appearing in Java are that the WPILib Java plugin is installed and that the project has the FRC nature and the Java nature. If the .project natures section on a "broken" computer matches that from a working computer (make sure the .project you are looking at is the one that is actually loaded in Eclipse, the project properties can point you at the disk location), an exception that is unloading the plugin is my only idea as to what's happening.

An uncaught exception thrown by the plugin any time between Eclipse launch and when you try to deploy will cause the plugin to unload and the menu item will not appear. If you want to post the Eclipse log (from WORKSPACE/.metadata/.log) I can take a look and see if there is anything suspicious in there.

docprofsky, 2018-02-24:
@pauldennis2 Your issue seems most like this line is missing from your `.project` file. Manually adding it and restarting Eclipse will probably fix it.
https://github.com/TeamCautionRobotics/2018-FRC-robot/blob/master/.project#L16

If not, you may want to try reloading the project (delete it from Eclipse without checking the "delete from workspace" checkbox then running File->Open Projects from File System..).

If that does not work, then try reinstalling Eclipse and the WPILib plugin. This usually fixes these problems.

ThadHouse, 2018-05-13:
With the switch to VS Code for the 2019, this issue is OBE. Closing.

closed 2018-05-13 by ThadHouse