wpilibsuite/allwpilib #951: ArrayIndexOutOfBoundsException in Command Scheduler

Javaru, 2018-02-19:
While a "While Held" command is executing, the following exception frequently happens. It causes the Robot to crash and restart the robot code. This is with WPILib v2018.3.3

```
ERROR  1  Unhandled exception: java.lang.ArrayIndexOutOfBoundsException: 1  java.lang.Thread.run(Thread.java:748)
	at edu.wpi.first.wpilibj.command.Scheduler.lambda$initSendable$0(Scheduler.java:339)
	at edu.wpi.first.wpilibj.smartdashboard.SendableBuilderImpl.updateTable(SendableBuilderImpl.java:86)
	at edu.wpi.first.wpilibj.livewindow.LiveWindow.updateValues(LiveWindow.java:282)
	at edu.wpi.first.wpilibj.IterativeRobotBase.loopFunc(IterativeRobotBase.java:227)
	at edu.wpi.first.wpilibj.TimedRobot.lambda$new$0(TimedRobot.java:30)
	at edu.wpi.first.wpilibj.Notifier.lambda$new$0(Notifier.java:97)
	at java.lang.Thread.run(Thread.java:748)

Error at java.lang.Thread.run(Thread.java:748): Unhandled exception: java.lang.ArrayIndexOutOfBoundsException: 1
	at edu.wpi.first.wpilibj.command.Scheduler.lambda$initSendable$0(Scheduler.java:339)
	at edu.wpi.first.wpilibj.smartdashboard.SendableBuilderImpl.updateTable(SendableBuilderImpl.java:86)
	at edu.wpi.first.wpilibj.livewindow.LiveWindow.updateValues(LiveWindow.java:282)
	at edu.wpi.first.wpilibj.IterativeRobotBase.loopFunc(IterativeRobotBase.java:227)
	at edu.wpi.first.wpilibj.TimedRobot.lambda$new$0(TimedRobot.java:30)
	at edu.wpi.first.wpilibj.Notifier.lambda$new$0(Notifier.java:97)
	at java.lang.Thread.run(Thread.java:748)

WARNING  1  Robots should not quit, but yours did!  edu.wpi.first.wpilibj.Notifier.lambda$new$1(Notifier.java:108)
Warning at edu.wpi.first.wpilibj.Notifier.lambda$new$1(Notifier.java:108)14:36:00.177 : Robots should not quit, but yours did!
ERROR  1  The loopFunc() method (or methods called by it) should have handled the exception above.  edu.wpi.first.wpilibj.Notifier.lambda$new$1(Notifier.java:109)

Error at edu.wpi.first.wpilibj.Notifier.lambda$new$1(Notifier.java:109): The loopFunc() method (or methods called by it) should have handled the exception above.
WARNING  0  [Phoenix] Shutdown cleanly  
********** Robot program starting **********
```

PeterJohnson, 2018-02-19:
Do you have multiple threads of execution in your code?

Javaru, 2018-02-20:
Yes we do.

ThadHouse, 2018-02-20:
All of the code in the command infrastructure is not multi threading safe. It only properly works with a single thread, and trying to run and cancel commands from multiple commands, is unsupported, and would definitely cause this issue given enough time. I would recommend moving all of your command based code to a single thread and seeing if you detect this error again.

Javaru, 2018-02-20:
Thanks. I'll take a look at the code. We don't call any commands on any threads other than the main thread of execution.  The other threads are primarily some utility threads. Nevertheless, perhaps one is interfering somehow. I'll turn them off and see if the issue is still present.

AustinShalit, 2018-05-30:
Closing this issue for now.  If the problem reappears please open with more information.

closed 2018-05-30 by AustinShalit