wpilibsuite/allwpilib #954: Inconsistent Time Unit in PIDController 'period'

Aric5301, 2018-02-24:
The Time unit of the 'period' variable in PIDController is implemented as seconds in the code. (Default value is .05).
In the documentation, it does not specify which time unit is to be used when this value is set by the user, but does say that default is 50 ms.
This is confusing and not intuitive beacause from the documentation it is not clear whether the methods that set this value get a number representing miliseconds or seconds.
Whatever the correct time unit is, it should be consistent through both code and documentation.

Best Regards,
Aric Radzin,
Lead Programmer,
BumbleB #3339

virtuald, 2018-02-24:
Feel free to make a PR fixing the documentation. :)

Aric5301, 2018-02-24:
I wish I could. but unfortunately I don't know to submit a PR...

virtuald, 2018-02-25:
PR = Pull Request

Some information about submitting a pull request can be found at https://help.github.com/articles/creating-a-pull-request/

PR #1305 (2018-09-12): PIDController: Mention unit for 'period'
Fixes #954
closed 2018-09-12 by PeterJohnson