wpilibsuite/allwpilib #955: SendableChooser.setName breaks updating the chosen value

docprofsky, 2018-02-24:
Calling the `setName()` function on a SendableChooser results in `SendableChooser.getSelected()` not updating to reflect the currently selected value.

We have found this issue in the 2018.3 and the 2018.1 or 2018.2 version of the WPILib Java plugin.

Here is a simple example that demonstrates this bug.

[image]

```Java
package org.usfirst.frc.team1492.robot;

import edu.wpi.first.wpilibj.TimedRobot;
import edu.wpi.first.wpilibj.smartdashboard.SendableChooser;
import edu.wpi.first.wpilibj.smartdashboard.SmartDashboard;

public class Robot extends TimedRobot {
    private SendableChooser<Integer> chooserA = new SendableChooser<>();
    private SendableChooser<Integer> chooserB = new SendableChooser<>();

    @Override
    public void robotInit() {
        chooserA.addDefault("oneA", 1);
        chooserA.addObject("twoA", 2);
        chooserA.addObject("threeA", 3);
        chooserA.setName("chooserA name");
        SmartDashboard.putData("putData A", chooserA);

        chooserB.addDefault("oneB", 1);
        chooserB.addObject("twoB", 2);
        chooserB.addObject("threeB", 3);
        SmartDashboard.putData("putData B", chooserB);
    }

    @Override
    public void robotPeriodic() {
        SmartDashboard.putString("selected item A", chooserA.getSelected().toString());
        SmartDashboard.putString("selected item B", chooserB.getSelected().toString());
    }
}
```

PeterJohnson, 2018-02-24:
Confirmed.  The issue is that the setName() causes the SendableChooser to also get added to LiveWindow, and SendableChooser isn't smart enough to handle being put into two different tables (the same issue would occur if you call SmartDashboard.putData() twice on the same chooser with different names).  For now, the workaround is to simply not call setName().

docprofsky, 2018-02-24:
How does `setName()` result in the Sendable Chooser getting added to the LiveWindow, since `setName()`, implemented by `SendableBase`, only

PeterJohnson, 2018-02-25:
SendableBase's are automatically added to LiveWindow by the SendableBase constructor.  However, LiveWindow.updateValues() (which is called from the main periodic loop in IterativeRobotBase) only actually initialize and add Sendable values to the LiveWindow if they have a name set.

One possible fix would be to have the SendableChooser constructor tell the SendableBase constructor not to add it to LiveWindow.

PeterJohnson, 2018-03-04:
Fixed by #964.

closed 2018-03-04 by PeterJohnson