wpilibsuite/allwpilib #957: Where to find the latest Roborio image?

SUPERCILEX, 2018-02-25:
Can you guys _please_ host a maven or something with all the Roborio images? This is the same problem every year: I have no idea where to get the latest image. It's v18 by now right? The imaging tool can only find v17.

PS: I have the 2018.1 toolsuite (which BTW needs a permalink since you guys always use the wrong year in the documentation).

Kevin-OConnor, 2018-02-25:
Nope. Images are always bundled with the NI Toolsuite. There is no 2018v18 as of today.

closed 2018-02-25 by Kevin-OConnor

PeterJohnson, 2018-02-26:
The RoboRIO images are distributed by National Instruments.  2018v17 is the latest image, and is distributed as part of the 2018.1.0 NI update (also the latest one).  The latest NI update can be found by going to http://ni.com/first, clicking on "Download 2018 FRC Software", clicking on "2018 FRC Update Suite", and then clicking on "Click Here to Download".  This takes you to a page which shows 2018.1.0 as the latest version of the update suite.