wpilibsuite/allwpilib #958: Compressor operation disables other pneumatics events

merritt5585, 2018-03-01:
Per the documentation for wpilibj, the Compressor class does not need to be instantiated explicitly but will be implemented to run the compressor with any instantiation of a solenoid class. This appears to work as described, but during the compressor run time, other pneumatics commands are disabled which is unexpected. 

Is this normal operation?

AustinShalit, 2018-03-01:
Can you please post a code example or your robot code so we can reproduce this?

AustinShalit, 2018-03-01:
For reference: https://www.chiefdelphi.com/forums/showthread.php?t=162777

merritt5585, 2018-03-01:
Austin:

The code in the post at your reference link is what I believe to be the relevant code from our robot. If you need more than that, please advise.

Thanks, Merritt

AustinShalit, 2018-03-01:
Can you please post a zip here of your entire robot project?  Just guessing based on the information I have so far: only one command can "use" a subsystem at a time.  Because your pneumatics are one subsystem, only one command can execute at a time.

merritt5585, 2018-03-01:
I'll upload a zip later this afternoon. All the code is on the laptop at
the school.

merritt5585, 2018-03-01:
[powerup2018.zip](https://github.com/wpilibsuite/allwpilib/files/1773145/powerup2018.zip)
[TestPneumatics.zip](https://github.com/wpilibsuite/allwpilib/files/1773147/TestPneumatics.zip)

First zip is an Eclipse archive of our entire robot code
Second zip is an Eclipse archive of a simple program that only contains code for a pneumatics subsystem set up to run a single air cylinder via a double solenoid.

merritt5585, 2018-03-01:
Sorry that I hit the wrong button when posting the zip files.

SamCarlberg, 2018-03-01:
Austin is probably correct. You have grouped logically unrelated objects into a single subsystem.

You can fix this by putting the unrelated solenoids into separate subsystems - one for the gripper, and one for the lift. Any command that uses the lift solenoid should require the lift subsystem, and any command that uses the gripper solenoid should require the gripper subsystem.

As a general rule, subsystems reflect _physically cohesive_ parts or mechanisms on the robot. Having a subsystem containing all the pneumatics makes about as much sense as having a single subsystem for all your motors.

`AutoDrive` misuses `requires`:  it declares a requirement on the pneumatics "subsystem", but does not actually use it. This prevents other commands (for example, `Grab` and `Lift`) from running at the same time (more specifically, running those commands would cancel a running `AutoDrive` - probably not the behavior you want)

So to recap:

- Create new subsystem classes for the lift and the gripper
  - `Grab` and `Release` should require the gripper subsystem
  - `Lift` and `Lower` should require the lift subsystem
- Remove unneeded `requires` statements

---

If the problems still persist after fixing the subsystem and command logic, or if they _only_ occur when the compressor is running, then something else is going wrong as well

merritt5585, 2018-03-02:
Thanks for the info.

First, the extra reuqires in the "AutoDrive" command was a cut/paste error
intended for one of the other commands and was inadvertently left in the
AutoDrive Code. This happened after the export of this archive and is not
in the current code for the full robot. Sorry for the confusion.

The problem we have is that wpilib's automated operation of the compressor
when instantiating the DoubleSolenoid class prevents any of the pneumatics
commands from operating while the compressor is running. When the
compressor is not running, all pneumatics commands work perfectly. We
hadn't seen the potential conflicts among pneumatics commands related to
the single pneumatics subsystem because we've not needed to run multiple
pneumatics commands at the same time.

In case of future need, we started the modification to use two subsystems
today but ran out of time and will finish next meeting. I don't have much
hope that the multiple subsystems will fix the problem with the running
compressor blocking the commands.

Merritt

merritt5585, 2018-03-06:
We've resolved this problem by instantiating a compressor in our code and then checking to see if it is in closed loop mode prior to running pneumatics commands. If it is in closed loop, we take it out of closed loop to stop the compressor, run our pneumatics command and then put the compressor back in closed loop. 

This allows us to run commands without them being blocked by the compressor. It isn't an ideal solution, but the robot is operating as it should so we're happy. 

I'm closing the issue since there seems to be no interest in addressing this in the WPI code. 

Thanks, Merritt

closed 2018-03-06 by merritt5585