wpilibsuite/allwpilib #962: Provide way to suppress various warning messages

PeterJohnson, 2018-03-03:
When testing it's quite common to e.g. not have a joystick connected.  Providing a way for users to suppress these warnings would be a useful feature to enable users to focus on any real issues in their code.

AustinShalit, 2018-03-05:
>Is a workaround possible to disable the joystick warning methods using reflection to modify the DriverStation singleton?

Sure but that is too ugly for us to recommend in the documentation.  If you want to do that, set the private [`m_nextMessageTime`](https://github.com/wpilibsuite/allwpilib/blob/master/wpilibj/src/main/java/edu/wpi/first/wpilibj/DriverStation.java#L69) field to a very large value like `Double.MAX_VALUE`.

Starlight220, 2020-03-14:
Is this still relevant?

Daltz333, 2020-03-15:
Yes

Starlight220, 2020-03-25:
The warning is only printed if the button is queried. Therefore I think the warnings should not be suppressed.

Daltz333, 2020-03-25:
Eg. The use case would be simulation and testing. Maybe suppression isn't the solution.

Starlight220, 2020-03-25:
Perhaps a suppressDisconnectedWarning flag on each Joystick object?

RobinsonZ, 2020-04-15:
I think it should be possible to suppress all warnings and console output, even loop time overruns. It should be at least somewhat tricky/unintuitive to suppress them so that people aren't cavalier about it (kind of like visiting a page with an invalid/unsigned SSL cert in Chrome), but if you know what you're doing there should be a solution that doesn't involve re-implementing classes.

For unplugged joysticks, what if that warning appeared once a program (or once every time the joystick state changed)? I don't really think it's necessary to spam the logs for that sort of thing. If a joystick is unplugged that was necessary for whatever a user is doing, they will probably find out soon enough anyway and don't need a hundred messages in the console.

sciencewhiz, 2021-01-03:
Is this fixed by #2845?

Starlight220, 2021-01-03:
I think so.

closed 2021-01-03 by Daltz333