wpilibsuite/allwpilib #966: Request a mirror site for doc in Mainland China

moesoha, 2018-03-06:
For some reason, accessing the official documentation site `https://wpilib.screenstepslive.com`  is really difficult in Mainland China.

[image]

According to the image, we can find that the DOM takes 12s to load and some images are failed to load. That's really normal to the foreign websites here. Someone will download the PDF version from the website and share it for Chinese teams. Obviously viewing the PDF instead of the doc site is the worst solution.

I want to set up a mirror site in Mainland China. Can I get the data and the site program to host a mirror? Or can I just set up a reverse proxy mirror?

Kevin-OConnor, 2018-03-06:
ScreenSteps Live is a service we use that is generously donated by a supplier. We do not have the rights or access to the code to provide for a mirror. I will look into whether there is anything we can do.

moesoha, 2018-03-07:
OK. I will try to host a reverse proxy service in China.

moesoha, 2018-03-12:
Mirror site in Mainland China is up. Speed is nice. https://wpilib.frc.lohu.info

closed 2018-03-12 by moesoha