wpilibsuite/allwpilib #971: Provide user feedback in SendableChooser implementation

virtuald, 2018-03-13:
Users tend to not trust SD/Shuffleboard/et al because it says one thing but there's no guarantee that the setting was actually received on the other side. However... https://www.chiefdelphi.com/forums/showpost.php?p=1745272&postcount=10 inspires me.

We attach a listener on the server side that flips a value on the server side, and shuffleboard shows a checkbox when the flipped value arrives back at its side (or maybe we use a sequence number, or whatever...). This would allow users to build more trust in the system.

PeterJohnson, 2018-07-30:
Fixed on the robot side by #1217.

SamCarlberg, 2018-08-09:
Fixed on Shuffleboard size with wpilibsuite/shuffleboard#497

AustinShalit, 2018-09-07:
Looks like this can be closed.

closed 2018-09-28 by PeterJohnson