wpilibsuite/allwpilib #975: PIDSourceType state should be contained by a PIDController object not a PIDSource object.

apozharski, 2018-03-21:
# Overview
The SetPIDSourceType method of PIDController breaks the tenets of OOP by allowing a PIDController object to change the state of the PIDSource object it uses. 
# Why is this a Problem?
This isn't only an issue for the "purist" in me but it also causes difficult to diagnose bugs and makes implementing certain control loop patterns difficult. An example of such a case is when there is a need for both a velocity and position controller based on the same encoder. As the PIDSource can not trivially be in both the `kDistance` and `kRate` modes it means that extra work needs to be done to allow this to be done. 
# Proposed Solution
Change the PIDSource interface to contain two methods, one to get rate and one to get distance. The 
`PIDSourceType` local variable should be moved to the PIDController class and request one or the other from its PIDSource depending on thatflag. If a PIDSource is not capable of either kind of behavior there can be several options to handle it:
* Provide dummy values (probably bad as it is hard to debug
* Throw an exception which will be handled and sent to the output by the PIDController.

calcmogul, 2018-03-21:
This change makes sense to me. We handle runtime programmer errors like setting an invalid PIDSourceType by throwing an exception in Java and calling wpi_setGlobalError() with checks of StatusIsFatal() in C++.

apozharski, 2018-03-21:
I think there is a clear benefit as this will allow for ease of use in several applications as well as removing the possibility of nasty bugs (i.e. changing the source type without understanding what that does, or doing it multiple times in several commands with possible interruptions causing unexpected results).  

I understand the desire to not break the API, however this feels like it was a design flaw of the PIDController class from the beginning.

calcmogul, 2018-03-21:
I agree. PIDController also doesn't support composition, like summing their outputs for a single actuator (e.g., position and heading controllers that drive the same two drive sides) or wrapping one controller around another such that the outer controller drives the reference of the inner controller (e.g., an outer position controller driving an inner velocity controller that actually drives the motor). I have some ideas for the latter, but I'll have to think about the former more. I'll file another issue.

PeterJohnson, 2019-07-30:
Fixed by #1300.

closed 2019-07-30 by PeterJohnson