wpilibsuite/allwpilib #980: Only deliver gameData in autonomous mode

virtuald, 2018-03-23:
If the robot isn't in autonomous mode, then it should return empty string. This will ensure a consistent experience on and off the field.

AustinShalit, 2018-03-24:
While game data is only sent at the start of autonomous this year, we have no indication from FIRST that this will be the case in years to come.  This is probably best implemented in the driver station rather than WPILib.

gerth2, 2018-05-07:
We'd like to keep the ability to calculate autonomous paths (or react in some way) as soon as the game data data comes in. If data is by rule available before autonomous starts, it would not be good to lose the ability to act on it.

JefferMC, 2018-05-07:
The data is just as valid in Teleop mode.  Why would you hide useful information?

gerth2, 2018-05-08:
> That's fair.
>
> The core problem I would like solved is that right now the DS/wpilib do
> not provide a consistent experience for gamedata on and off the field,
> resulting in confusion and dead robots.

Yupyup, I think there was some CD thread on it I'll have to go look up.
Basically, some mechanism to know when the data was stale vs. current for
the present match would be desirable, along with some sort of assurance
that the data will arrive in a specific timeframe or the match gets
replayed.... I think that's dipping more into the driverstation and FMS
implementation than WPIlib though.