wpilibsuite/allwpilib #985: NullPointerExcept

jordanbancino, 2018-03-28:


AustinShalit, 2018-03-28:
Your robot code is throwing the exception because the game data has not been received yet.  Please update to the latest WPILib (2018.4.1).  This introduced a change that instead of returning null when there is no game data it will instead return an empty string.  Also, please **only** check for game data in `autonomousInit`.  Game data is not sent to the DS and then to the robot until the match is started.  From the first section of the [Game Data Details](https://wpilib.screenstepslive.com/s/currentCS/m/getting_started/l/826278-2018-game-data-details) document:
>Plate assignment data is provided to each robot (via the Driver Station) by the FMS just prior to the start of each match. The data will be sent after the "3...2...1...Go", and a short animation of the plate lighting will illustrate the randomization to ensure robots have time to receive and react to the data. To ensure your program properly responds to the data, you should either poll the data in a fairly fast loop while Disabled (~20ms) or retrieve the data once on the transition to Autonomous Enabled.

Please read the entire Game Data Details document for more information: https://wpilib.screenstepslive.com/s/currentCS/m/getting_started/l/826278-2018-game-data-details

jordanbancino, 2018-03-28:
@AustinShalit My robot is not throwing an exception. The exception occurs in the RobotBase, thus preventing my code from even starting. I am properly catching any exceptions my code could throw.

jordanbancino, 2018-03-28:
@AustinShalit Also, we did have a driver station connected, so this is not a duplicate of #895 . We played a match where the FMS simply did not send any data to us, shutting down our robot during  robotInit. To be clear, we are not making any calls to the driver station in robotInit, so our code is not the issue here. I will take a look at the stack trace, and see what more information I can find.

AustinShalit, 2018-03-28:
I'd like to take a look at your code.  Can you put it somewhere I can access it and post a link here?

jordanbancino, 2018-03-28:
@AustinShalit Yes. I will put it here as soon as I can, along with the error stack trace from the driver station.

jordanbancino, 2018-03-28:
Ok. Here is the code I am using:
```java
try {
    gameData = DriverStation.getInstance().getGameSpecificMessage();
    ourGameData = gameData.substring(0, 2);
catch (Exception e) {
    ourGameData = "none";
}
if (ourGameData.equals("none")) {
   // Run default auton mode because no game data was aquired.
}
```
Originally, the `catch` block only caught `NullPointerException` and `StringIndexOutOfBoundsException`, but we changed it to just catch all exceptions.

This should work, but a `NullPointerException` is thrown when no game data is received, indicating that the exception is being thrown in a layer above what our code is running at. Unfortunately, I lost the Stack trace, and I don't have the resources to replicate it, but I'm pretty sure the exception occurs in `RobotBase`

AustinShalit, 2018-03-28:
You can find the stack trace in the DS log viewer.

Open the log viewer and on the left hand side select the match that gave you trouble.  The exception will be reported in the "Event List" tab at the top of the display.  Photos and more instructions here: https://wpilib.screenstepslive.com/s/currentCS/m/driver_station/l/144980-driver-station-log-file-viewer

jordanbancino, 2018-03-28:
That's not what I meant. I don't have access to our driver station PC.

virtuald, 2018-03-28:
@jordanbancino you need to catch `Throwable` if you're using that approach. Catching Exception won't catch `NullPointerException`.

With that said, it should never be null. If it is, you're using an old version of WPILib.

jordanbancino, 2018-03-28:
Ok. I'll test catching `Throwable`, and see if that works. I know we got null once, but we just updated our roboRIO and all of our dev computers about a week ago, so we are on the latest WPILib.

virtuald, 2018-03-28:
Keep in mind that (as Austin mentioned above) if you retrieve game data anytime other than `autonomousInit/autonomousPeriodic`, then an exception will be thrown. So don't do that, and you won't have to worry about whether an exception was thrown or not. :)

jordanbancino, 2018-03-28:
@virtuald Yeah, we retrieve game data in `autonomousInit`. 
@AustinShalit Ok. I modified my code:
```java
try {
    gameData = DriverStation.getInstance().getGameSpecificMessage();
    ourGameData = gameData.substring(0, 2);
catch (Throwable e) {
    ourGameData = "none";
}
if (ourGameData.equals("none")) {
   // Run default auton mode because no game data was aquired.
}
```

But, I just tested it, and it results in the following error, and our robot dying:
```
Error 1 Unhandled exception: java.lang.NullPointerException
edu.wpi.first.wpilibj.RobotBase.main(RobotBase.java:263)
```

So, we are still getting a `NullPointerException` when we don't get data. I'll try :

AustinShalit, 2018-03-28:
Please post your entire robot program to a place we can look at it.  Something is throwing a null pointer exception in robotInit.

jordanbancino, 2018-03-29:
What would be causing that in `robotInit`? We aren't retrieving any information, only setting up some `Talon SRX`'s and defining constants. We only get this error when we get `null` from the Driver Station (which  occurs in `autonomousInit`), which has actually happened to us in a match.

333fred, 2018-03-29:
To be clear, this is line 263 in RobotBase: https://github.com/wpilibsuite/allwpilib/blob/master/wpilibj/src/main/java/edu/wpi/first/wpilibj/RobotBase.java#L263. Something else is very wrong if you've managed to make a static method null pointer.

AustinShalit, 2018-03-29:
Do you ever see the message: "Robots should not quit, but yours did!"?

jordanbancino, 2018-03-29:
@AustinShalit Yes.

virtuald, 2018-03-29:
.. what was it?