wpilibsuite/allwpilib #997: free() method not consistently implemented

AustinShalit, 2018-04-27:
SendableBase has a method `free()` that most classes that extend it override.  Gyro also has a `free()` method that does a similar thing.  It does not seem like this is the right abstraction we want.  Maybe we need a `Hardware?` interface that defines the free method.

AustinShalit, 2018-05-08:
It was mentioned that maybe we should use AutoCloseable here so we can use hardware in a try with resources statement.  Not sure of the C++ equivalent.

ThadHouse, 2018-05-11:
There is no equivalent for C++, it just uses destruction (which we already have hooked up properly). I implemented this in #1048, however it adds a large amount of warnings to user programs we might want to think about.

PR #1048 (2018-05-23): Implements Closeable for types, replacing free()
Closes #997
closed 2018-05-23 by PeterJohnson