wpilibsuite/allwpilib #1031: Are SendableBase constructors not called intentionally?

virtuald, 2018-05-05:
It appears that in a lot of wpilibj objects they don't call constructors that call the SendableBase constructor ([which then calls LiveWindow.add](https://github.com/wpilibsuite/allwpilib/blob/b7ea481bf98e91bf665f1273d7428b19ba0d2b39/wpilibj/src/main/java/edu/wpi/first/wpilibj/SendableBase.java#L33)). Is that intentional?

* If it is, then for consistency and obviousness' sake I would advocate that the functionality in SendableBase be extracted to a separate function and the classes that need it should call it
* If it isn't, then the base constructors should be called in all wpilib classes always (or a comment added that it was intentionally not called)

It seems it would be more consistent to always call the base constructor on every object (I feel this is a best practice anyways?).

Reference: https://github.com/robotpy/robotpy-wpilib/pull/463

AustinShalit, 2018-05-05:
Because SendableBase has a no arg constructor, it is automatically called when a subclass constructor is called.  The no arg constructor functionality does add the sensor to the LiveWindow. https://github.com/wpilibsuite/allwpilib/blob/b7ea481bf98e91bf665f1273d7428b19ba0d2b39/wpilibj/src/main/java/edu/wpi/first/wpilibj/SendableBase.java#L20-L36

https://docs.oracle.com/javase/tutorial/java/IandI/super.html
>Note: If a constructor does not explicitly invoke a superclass constructor, the Java compiler automatically inserts a call to the no-argument constructor of the superclass. If the super class does not have a no-argument constructor, you will get a compile-time error. Object does have such a constructor, so if Object is the only superclass, there is no problem.

virtuald, 2018-05-05:
Learn something new every day. Thanks!

closed 2018-05-05 by virtuald

AustinShalit, 2018-05-05:
Something I just thought of is that we could make this explicit via a [styleguide rule](https://pmd.github.io/pmd-6.3.0/pmd_rules_java_codestyle.html#callsuperinconstructor).  Thoughts on doing that @calcmogul?

virtuald, 2018-05-05:
That styleguide rule would make it easier to maintain and validate RobotPy source.

AustinShalit, 2018-05-05:
Ok.  Once I am done reenabling checkstyle & pmd, I'll open a PR to add that rule for further discussion.