wpilibsuite/allwpilib #1035: Exposing CAN interface functionality in API

gerth2, 2018-05-07:
What would it take to make the RIO's CAN bus available for sending/receiving data in the C++ & Java API's?

I'm thinking if users want to implement custom devices & drivers in the future. I know from the Java side there's some JNI interfaces, but nothing I've found documented for interacting with the CAN bus. I haven't dug into the C++ side, maybe there is more already?

JacisNonsense, 2018-05-07:
This functionality is possible for java using the CANJNI, and in C++ using HAL/CAN,h. It is fairly low level but does expose the correct functions for sending and receiving over the CAN bus. 

It might be worth trying to implement some wrapper classes, though, to make things a bit easier on the user side.

ThadHouse, 2018-05-07:
As you said, the JNI interface is already there, and there have been a few teams that have used it in the past. But I do agree it is not well documented. In good news. I am actually working on an easier to use CAN API that should be added sometime this summer. So  keep a look out!

closed 2018-06-02 by calcmogul