wpilibsuite/allwpilib #1037: Mocks of WPI classes

brennonbrimhall, 2018-05-07:
From https://www.chiefdelphi.com/forums/showpost.php?p=1766121&postcount=7:
> I’d like to see mocks of the different WPI classes so teams could easily unit test different subsystems.

It seems to me that the easiest way to accomplish this would be in the HAL.  My preference would be to implement this completely in Java for the Java teams and in C++ for the C++ teams.  This would allow those teams who use Java to not have to load in a shared object at runtime.  

I'd also like to see something similar for integration testing - some mock of the Driver Station, complete with joystick support.

This is potentially related to #922 and #1024.

ThadHouse, 2018-05-07:
We have a mock HAL already in C++, and recently we added the API's to be able to talk to this from Java well. There are already shared libraries needed to be loaded for NT, so adding another isn't any more difficulty. We are working on the documentation for these APIs and how teams will be able to use them, and that is one of our big goals this summer to get things better. 

Here is the PR adding the user facing APIs for both Java and C++
#1002

brennonbrimhall, 2018-05-07:
Looks great!  I'm more than happy to assist in the documentation process.  Just let me know.

brennonbrimhall, 2019-01-18:
Did this ever get documented @ThadHouse?

ThadHouse, 2019-01-18:
No, we had to focus more time on the vscode and gradlerio change. But the functionality is merged in and ready to go. For c++ look in the simulation header folder, for Java it's in the edu.wpi.first.hal.simulation package.