wpilibsuite/allwpilib #1041: Use Builder pattern for Motor Controllers

twinters007, 2018-05-09:
When creating simple wpilib drivetrains, if you want to configure motors you have to create a motor object, configure it, and pass it in, rather than just configure it inline without creating a reference to it. I think it would be more ideal to use the builder pattern since it is designed for this sort of application. 

You'd be able to do 

```
DiffDrive drive = wpilib.drive.DiffDrive( (new Talon(1)).setInverted(true).setSafetyEnabled(false), 
                                         (new Talon(2)...
```
which is nice than creating an object for each talon.

calcmogul, 2018-05-10:
I'm not sure your example there is really a shining example of the builder pattern; there's multiple nested non-native object constructions on the same line. Also, this would leak memory in C++.

twinters007, 2018-05-12:
Why would it leak (not arguing, just wondering)

calcmogul, 2018-09-19:
DifferentialDrive takes references to SpeedControllers in the constructor and doesn't take ownership of the objects. If objects are allocated with new like that, the pointers won't be deleted by anyone later, and thus the memory will leak.

It's been a while, but I think we decided against this internally. We'd have to make functions like setInverted() and setSafetyEnabled() return SpeedController instead of void, and we're not fans of making that kind of change for new syntax for the user that isn't that compelling anyway.

closed 2018-09-19 by calcmogul