wpilibsuite/allwpilib #1046: Should MecanumDrive extend DifferentialDrive?

AustinShalit, 2018-05-10:
Is a mecanum drive a kind of differential drive?

twinters007, 2018-05-10:
Since the wheels operate separate from one another, technically yes.

calcmogul, 2018-05-10:
It is not (not even in the technical sense above). Mecanum drive is a holonomic drivetrain, which means it can move omnidirectionally. Differential drive cannot do this, so it is non-holonomic.

twinters007, 2018-05-10:
I would say holonomic is a sub-class of differential, because you can do everything you can do with differential with mecanum

bradamiller, 2018-05-10:
I have to agree that they’re not the same. Example: tricycles have three wheels and move, airplanes have 3 wheels and move, so an airplane is-a tricycle. Also mecanum drive must have 4 wheels.

AustinShalit, 2018-05-10:
Ok.  I am convinced.  Thank you @calcmogul & @bradamiller!

closed 2018-05-10 by AustinShalit