wpilibsuite/allwpilib #1072: lo-fi simulation master ticket

virtuald, 2018-05-17:
The goal for this ticket is to coordinate the work needed to have simulation (such as RobotPy's [pyfrc](https://github.com/robotpy/pyfrc) or @pjreiniger 's [SnobotSim](https://github.com/pjreiniger/SnobotSim) available that students can easily use without needing to do much (if any) work for basic logical testing.

Currently @jwhite66 is working on getting gazebo support working, but while Gazebo is related it is different and is NOT something intended to be discussed in this ticket.

The envisioned workflow as I understand it is that the difference between the 'normal' robot code and the simulation robot code is that the robot code will be linked to a HAL that provides simulation support and to a frontend that will provide a UI for a user to interact with.

One thing that I believe is important is that 'physics' support needs to be provided that allows teams to provide code that allows the simulation to respond like a real(ish) robot. WPILib should provide some example helpers and motor models (and projects) for doing various types of simulation (such as what snobot sim provides), but there also need to be easy to use hooks that allow users to create their own motor models or whatever.

There have been discussions about an appropriate frontend -- HTML, JavaFx, even QT? This should be debated and decided.

In the past it was discussed whether the simulation frontend should be standalone and communicate with the underlying program via NetworkTables (so Shuffleboard could be used, for example), or if the frontend should be linked to the robot code directly.

@ThadHouse has done a lot of work on the low level plumbing needed to make this work, and I believe already has example unit tests that use the simulated HAL which may be useful for someone interested in working on this stuff.

virtuald, 2018-05-17:
One potential outcome of this ticket would be the creation of a design doc submitted to the [design docs repo](https://github.com/wpilibsuite/design-docs) detailing the work that needs to be done.

jwhite66, 2018-05-17:
In addition to the Gazebo work, I also have two useful branches for this project.  One is minor fixes in the sample programs, so they can be run.  The other allows the simulator to be operated with a real driver station, which is handy.  I'll submit those PRs once I've made headway on the Gazebo patch set.

Open issues in my mind include:
  * gradlerio seems like the obvious place to put the smarts to assemble and launch the simulated program.  It has a start on it, but needs work.  At the same time, it's nice to be able to build and launch the sample programs under a simulator, so we end up duplicating the work.  Would it make sense to shift the sample programs out of wpilib and into gradlerio?

  * The CTRE libraries are not presently supported.  That is a hard problem, and likely to seriously reduce the value of any simulator.

school19, 2018-05-17:
If one of the goals is to have the lo-fi simulation is to have it behave as a real robot, then there also needs to be a way for teams to define what their robot looks like and what the moving parts are, but it was pointed out that a 3d simulation would require programmers waiting on a completed design. Would this be a top down 2d simulation then?

virtuald, 2018-05-17:
There's no need for teams to define what their robot looks like, only how it behaves. pyfrc is a top-down 2d simulation, and I would expect this to be similar.

pjreiniger, 2018-05-17:
Thad has done great work getting the low level pieces set up.  The next big piece is getting the glue that attaches motors to sensors, which includes to the physics simulation and providing an extendable way for third parties to hook into this.  Figuring out the best way to implement this should be priority number one.

I think providing the hooks to configure these simulations is an important part of this epic, but a story that can be kicked down the road quite a ways.  Same goes with having a standalone program that can be launched by `gradlew simulate`

virtuald, 2018-05-21:
We've had a good working relationship with CTRE in the past. My expectation is that the minimum required to make simulation work with SRX should be 'easy'. I believe Thad has already done some preliminary work experimenting with adding external vendor libraries to simulation, and as long as we make that *possible* and *easy* then I'm sure CTRE will implement the minimum required to get it working (eg... set/get motor percent, encoder set/get, etc).

The 'hard' part is simulation of the advanced control modes that the SRX offers. I'm less confident that it could be implemented this year because of time constraints... but @JCaporuscio or @ozrien could speak to that better than I.

pjreiniger, 2018-05-21:
I have together some mocks of thier stuff, but it would obviously be better if they had some control of it https://github.com/pjreiniger/SnobotSim/tree/dev-2019/ctre_override/src/main/native/cpp/ctre_cci_mocks

The missing piece is that it ties heavily into the snobotSim architecture, for which wpilib does not have a similar comparison yet

ozrien, 2018-05-22:
Some thoughts...
- I think simulating the CCI calls is the right way to go.  That way Phoenix-frc-lib/Phoenix-LabVIEW  should be identical to both simulation and real-world usecases.
- The simulation implem of CCI would likely take several milestones.  The first one would just be open loop modes only and naive constants for all signal inputs (temp/current/voltage/sensor values).  Second would be faking sensor values, bus voltage, and current as a function of motor output.  Third would be honoring all of the remote sensor and MPE features.  Even having the first milestone accomplished would be a step in the right direction.
- I'm not sure but I would hope having a simulation-CCI would help us (CTRE) regression test things.  But I'm not certain if that's true.  Maybe someone from WPI can confirm if how useful their simulation platform is for maintenance.
- We have no idea where simulation libraries go, or how to do them.  This will complicate our build process (have to build another binary) and installer (how to place more files on the PC).  So unless there is a clean example to play with, we won't be able to start. 
- If this ends up being something that just confuses our customers (because they select simulation when they want hardware or vice versa) than this will just hurt everyone involved. So I think we (CTRE) first would need to access how useful/risky simulation features are in WPILIB.
- What about LabVIEW?

ThadHouse, 2018-05-22:
Ill address the bullet points I can as by the number they are in your post.
1. Yes, simulating CCI calls is definitely the right way to go. 
2. Thats how we did it too.
3. One nice thing it enabled us to do is have unit tests that test the high level APIs. So when we build wpilib, it actually runs unit tests against production wpilib code, even on desktop, and just has mock HAL calls put in. So the support does help alot.
4. The build process is a little more difficult to set up. We end up building all of our binaries for multiple platforms to support these scenarios. We need to discuss this more, but the dependencies really should be coming entirely through maven, even for the roborio. It makes gradlerio much happier. I think we need to have this part of the discussion offline with a more limited group though.
5. With everything set up in gradlerio and vscode right, this is not an issue. With separate maven packages, gradlerio knows which target is requested and grabs the right one.
6. We've always had our simulation separate from LabVIEW, so I don't really have a good answer for that one.
7. Our 3rd party sim lib is the HAL. there is no library with the NetComm calls and the ChipObject calls in Sim. However, you would only end up with the HAL dependency for sim. It could still continue to be completely unneeded for the roborio dependency.

For the multiple platform builds, I've been doing a lot of work on getting a gradle side build much easier to set up. I cany start working on a PR into your public repo to give an idea of what it would look like. It would require no changes to the way the CCI is built currently, just some build.gradle changes, and it will just start working.

virtuald, 2018-05-23:
RobotPy only has simulation-based testing, and it's very effective at finding particular classes of bugs, and goes a long way towards ensuring that we don't accidentally break things.

pjreiniger, 2018-05-25:
Step 1: Simulating motors.

This is really a core part of the simulation process.  SnobotSim uses a similar version of the image below.  Each motor controller keeps track of its own position, velocity, acceleration, and current which can then be accessed by attached feedback sensors.  These values are are calculated using a motor model simulation, the most basic of which assumes a perfectly linear response of a motor with a max speed of N.  I stole some other models from the Cheesy Poofs, but those can come in later.

The implementations of the motor simulator can directly talk to the HALSIM functions and callbacks (or, CTRE can make their own implementation to talk to whatever they implement for simulation).

Thoughts on this being the groundwork for the low-fi simulation?

[image]

PS I guess Step 0 is figuring out the project layout.  I assume that this will be written in C++ so that C++ teams can run a headless version of the simulator so they can easily run profilers or whatever.

virtuald, 2018-05-26:
I do like the way that snobot has its simulation classes set up (which this seems similar), and this seems pretty reasonable.

What I often do when designing APIs is write out a notional example of whatever it is that I'm trying to do and try to simplify things until the example looks like what I want, then fill in the actual API from there. I think that would be a useful thing to do for this to see what pieces we need... but I think that's because it's not really clear to me what the code that the user implements looks like yet.

Clearly I'm biased, but I like [how pyfrc works](https://github.com/robotpy/examples/tree/master/physics/src), where the user just defines a PhysicsInterface and the simulation is able to magically find it and plug in the right pieces. I suspect for Java/C++ that perhaps anything simulation related would need to go in a special namespace/folder that way it doesn't get compiled in during robot deploy.

Also, would it make sense to bring a bunch of stuff over from snobot to here?

chuckb, 2018-11-27:
What I have to offer is not simulation per se, but I think it falls in line with the spirit of this ticket. And so instead of starting a new issue, I will document here. I believe that project template starting points from the the VSCode plugin, and/or examples, should not present impediments in either design hints or project setup mechanics to enable easy use of JUnit testing and Mockito mocking.

I have the following observations based on the work I have done on this subject:

1. In the latest revision (beta-2) I have tested, all external dependencies that would prevent the direct instantiation of commands and subsystems from a JUnit test are gone. In the 2018-4-1 build, that was not the case. One had to do:
```
    static {
        HLUsageReporting.SetImplementation(new HLUsageReporting.Null());
    }
```
I mention so as to keep stuff like this from happening in the future.

2. JUnit testing of command groups is possible with the use of the WPILib scheduler. However, the MockHardwareExtension.java is not available in the automated gradle download, and the scheduler will not instantiate without it. Performance of tests with the extension is not particularly great (given that HAL and friends is set up and torn down each test class). In my testing, and based on the coding style I am recommending, the HAL is not even really necessary, as I am using Mockito to mock all hardware as I see fit. The scheduler is absolutely is necessary to test command groups. Network tables is sometimes nice if you want to do system testing. I guess what I am saying is that having a really skinny environment for just the scheduler would be ideal.

3. The current beta-2 build is dragging down wpihaljni.dll built against the debug VCRT. I suppose that maybe makes sense because the project is building in debug mode, or the packages in maven are just all deployed that way at present because of beta. Either which way, it requires me to install Visual Studio 2015 just to test. Yuck. I would not want to have to do that for all my student workstations.

4. I built a simple scheduler to unit test commands. It therefore does not require all the HAL and friends overhead. Maybe this would be useful to include?

For reference, please see my hello world project with extensive test samples: https://github.com/Team997Coders/TestableRobot/tree/master

Further, I have a youtube play list that chronicles my journey to get this working for my team: https://www.youtube.com/watch?v=vmRFiF9hd2E&list=PL-u0oNZBsNMYXy51FJlCpRePpXzJ6XMU_

I will be exploring forking the plugin with the hopes of PR in the next few days.

ThadHouse, 2018-11-27:
One thing I do want to bring up @chuckb is we had the discussion about forcing injection into new commands rather then statics over the summer, and actually decided against it. We always have to think about the teams without mentors first, and having injection is more complicated for users without programming help or knowledge. All the templates are stored in this repo, not in the plugin or in GradleRIO, however a PR changing away from statics in the templates will likely not be merged. 

An example showing that off (not template) would likely definitely be accepted, and I personally think that would be super cool.

chuckb, 2018-11-27:
> One thing I do want to bring up @chuckb is we had the discussion about forcing injection into new commands rather then statics over the summer, and actually decided against it. We always have to [...]

Is my point 2 addressable and point 4 useful? If so, how can I help make it so?

Regarding templates, would it make sense to at least include testing bootstrap, like having a test directory set up, launch.test.json existing with the correct path to jniExtractDir so that test runner helpers work (they don't otherwise), correct bits to enable testing in build.gradle, and a correct .classpath so that tests will compile? All these niggly things make getting started with testing a chore.

One more thing I forgot: when I run a test with ```@ExtendWith(MockHardwareExtension.class)``` it runs fine from the command line. But when I run it through the test helper, it hangs (both run and debug). The test completes but I get no return. 

I suppose that even with the use of tight coupling, one could still run tests with ```@ExtendWith(MockHardwareExtension.class)``` specified. Correct? Mocking data through the HAL is not  something I know whether easy or hard.

ThadHouse, 2018-11-27:
The build.gradle file in Beta 2 should already be setup to enable testing and the correct classpath for testing. Is it not? The test directory I can add with a basic test class, can you create an issue in vscode-wpilib so I remember to do that?

The launch.test.json I'll look into, but last time I checked it required full path directories? Is that still the case? That was one of the reasons we don't have it, and I actually have an open PR to the vscode test extension to allow disabling those test helpers.

I don't know enough about Java testing to know why the @ExtendWith works from command line but not the test helper.

For at least beta 2, the HAL mock calls are there and can freely be used. Theres no docs on how to use them yet, but they are there and do work.

Yes, fork and PR here for the example works.

chuckb, 2018-11-28:
Yes and I will start a new project making sure I have the latest plugin and try again. And I will do the ticket.

Yes, I think it still does and I also put in a ticket to have them support the workspace macro. I had thought the plugin was doing the template instead of just copying, so it would have been easier to fill in.

It seems like some kind of bug with resource shutdown when called from the helpers. Just a guess. I can go further and see if I can find.

I'll look at HAL mocks and see if it is any easier (or better/required for certain scenarios) than using mockito.

pjreiniger, 2018-11-28:
[SnobotSim](https://github.com/pjreiniger/SnobotSim/blob/master/ReleaseNotes.md) does a lot of HAL mocking. My goal was to push more into wpilib, but my real job got quite out of control during the summer and fall and I couldn't find the time.

chuckb, 2018-11-29:
Before I send you a PR, is there a way for me to go about testing my added example? My guess is that the build boilerplate may not add JUnit nor Mockito.

chuckb, 2018-11-29:
So I have a testable robot example ready to PR. Phew, the style checks are thorough ;-)

Before I do, how can I do a final check that the VSCode plugin (which I have not touched...and can if I need to to get this to work) gens and runs as expected? I had the example pretty thoroughly tested, but precompile checks shook the jello.

ThadHouse, 2018-11-29:
If your testable code doesn't require any gradlerio changes, as long as it compiles and runs in allwpilib it would work in vs code. Testing that is a bit difficult and I need to make up some instructions. But open the pr and it's easy to check from there.

chuckb, 2018-11-29:
Silly question...is there a runner for examples? Running my example from manual constructed command line with all jars from ./build/allOutputs in the cp, but opencv is a dependency on CameraServerJNI and not found. I don't see the opencv jar anywhere in the hierarchy, so this leads me to believe there is a step I am missing.

AustinShalit, 2021-03-13:
With the addition of Glass and additional simulation features, I am going to close this ticket out. Please open a new ticket to track additional new features.

closed 2021-03-13 by AustinShalit