wpilibsuite/allwpilib #1083: Default source documentation style (XML vs JSD)

ozrien, 2018-05-22:
Currently WPILIB uses JavaDoc Style (JSD) for comment function headers.  I don't know what VSCode works best with.  VS Studio heavily embraces the Microsoft XML [style](https://docs.microsoft.com/en-us/dotnet/csharp/programming-guide/xmldoc/xml-documentation-comments).

Has anyone tested the intellisense features in VSCode to see if JSD works?

ThadHouse, 2018-05-22:
VSCode doesn't currently support any C++ style docs. Eclipses were not set up properly either, so they were not always working either. Java works fine with JavaDocs

PeterJohnson, 2018-05-22:
It’s not on the roadmap either.  The only thing talked about was xml, but I don’t think we want to switch until we know.  Plus does doxygen support the xml format? See  https://github.com/Microsoft/vscode-cpptools/issues/399

PeterJohnson, 2019-09-14:
VSCode supports Doxygen style comments (JavaDoc style).

closed 2019-09-14 by PeterJohnson