wpilibsuite/allwpilib #1097: Move wpilibc headers into WPILib/ folder

calcmogul, 2018-05-25:
The headers are currently polluting the system include directories. With the cRIO control system, we used to keep them in a WPILib folder with a `-I` directive into it so users didn't have to specify "WPILib/" in their #include directives. We could use the same approach to clean up our header organization, but also not break old code.

This will require adding a `-I` directive to GradleRIO's compiler invocations, but the header move won't break any users who are using that version of GradleRIO or later.

ThadHouse, 2018-05-25:
Note this isn't exactly as simple as adding that directive on the gradlerio side for other reasons, but we can definitely make it work.

PeterJohnson, 2018-07-27:
The headers were moved in #1200.  That PR also added shim headers for backwards compatibility.

closed 2018-07-27 by PeterJohnson